tuya-cloudcutter/tuya-cloudcutter #1: Release Day

tfrew-r7, 2022-03-24:
What day will this toolset be released?

closed 2022-03-25 by tjclement

btsimonh, 2022-03-25:
I guess I get the soldering iron out to put openbeken on the device which will arrive today, and light my kitchen tomorrow....  I suppose I can always test on my test device by returning it to the factory firmware. - and maybe test on the device which is arriving as it's not TOO difficult to extract after installation.

I take it you will want reports of success to add to the list?

Looking forward to the details next week!
br,

Simon

digiblur, 2022-03-25:
Looking forward to Tuesday with half the digiblurDIY Army in tow..

tfrew-r7, 2022-03-25:
Will the release include a dumper? Keep getting a CRC error trying to dump an N chip

khalednassar, 2022-03-25:
@tfrew-r7 have you tried https://github.com/khalednassar/bk7231tools ? There's a flag for this exact issue described in the README.

tfrew-r7, 2022-03-25:
Thanks, I will dump with that tool later today, should I email you the files?

digiblur, 2022-03-25:
> @tjclement - device is the one in this post: https://www.elektroda.com/rtvforum/viewtopic.php?p=19932845#19932845 full dump (less bootloader) is attached to the post. My other one (my true dev device) is a uk Calex LED strip, will use this to test the 'methods' of extracting the required :).
> 
> @digiblur - completely off topic - you have Tuya devices with Tasmota? - i'll need some help with Tuya testing... post an isuse at https://github.com/btsimonh/Tasmota/tree/Tuya if you feel you could help.

Sent you a message on that proejct branch.

btsimonh, 2022-03-25:
ref bk firmware dumping, note also my fork of the beken python stuff...  also does SPI from RPi if you stuff the bootloader...
https://github.com/btsimonh/hid_download_py

btsimonh, 2022-03-28:
tomorrow !!!

khalednassar, 2022-03-29:
> tomorrow !!!

_now!_

btsimonh, 2022-03-29:
Nice work!

_I'm going to assume that Tuya are looking at this thread...._
**To Tuya:**
For those of us who love the hardware cost, but hate the cloud requirement (and have a strong distrust of commercial closed source networkable firmware), it would be great if Tuya took this into account in their devices - 
I see two ways;
1/ enable local control by default, allowing people to block outgoing traffic, or run in an isolated LAN.
2/ enable local flashing with OSS firmware.  It may never be as featured as the Tuya ecosystem,  but that's our loss.
_Let's face it, if these were Russian originated, we would have been told to turn off all of our lights in the last few weeks..._