tuya-cloudcutter/tuya-cloudcutter #4: Devices bricking with ! in the ssid password

MoogleTroupe, 2022-03-29:
Tested both by passing in as an escaped parameter, and modifying the python to hardcode what's used. 

Device never joins the network, and upon trying a factory reset it starts up an AP that cannot be joined by phone/computer/cloudcutter tools.

MoogleTroupe, 2022-03-30:
Update on this, appears the devices doing this can be recovered by doing the 6 power on/off cycle a couple of times, fixes the ap mode not accepting connections anymore. ssid password with ! in it still causes the behavior.

MoogleTroupe, 2022-03-30:
Finding having a space in the ssid name is also causing this same type of behavior, maybe there is some encoding issue in the config payload for anything that isn't a letter or a number?

MoogleTroupe, 2022-03-30:
found the specific behavior when it goes into this fault and then back out-> power cycle 6 times. it slow blinks but is broken, wait a few minutes and you will see it hard lock up for a little while, then when it starts blinking again the ap is then working and you can run the exploit again.

khalednassar, 2022-03-30:
Think this needs a bit more debugging. I'll take a look at it when possible, thanks!

Also the command invocation with bash in docker is indeed causing grief, will take a look at that

MoogleTroupe, 2022-04-01:
unfortunately 3ed1a2b17f701682444818e781a8537f8bf17d20 doesn't seem to have fixed having spaces in the SSID name, "test wifi" "password" isn't working but "testwifi" "password" is. I haven't tested if ! is still broken or not.

khalednassar, 2022-04-01:
Just pushed 997d11167e0f4e497763bd36aa47983037605246 to deal with the bash escaping issues as well. In my testing, it seems to work fine with a password including `$ @ !` and spaces. Can you test that on your side?

khalednassar, 2022-04-02:
There were more issues related to spaces because of odd handling in tinytuya. Should be fixed with 9fa026be91a9866db8c123c95d98776dc08acf93

closed 2022-04-04 by khalednassar