tuya-cloudcutter/tuya-cloudcutter #10: WB8P dump from STITCH

tony-fav, 2022-04-04:
WB8P: https://developer.tuya.com/en/docs/iot/wb8p-module-datasheet?id=K9fwx4f89tvzd

STITCH Bulb: https://www.monoprice.com/product?p_id=42267

Used https://github.com/OpenBekenIOT/hid_download_py to dump default size and length 0x1EF000 

[Stitch_Bulb.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8406343/Stitch_Bulb.zip)

khalednassar, 2022-04-04:
Profile created in 808634da2cb994feb4b030f5de565763c597faf0. I couldn't pull the DPID schema from the dump (information about the device needed to get it is in 0x1EF000 and beyond), but I presumed it shouldn't be _too_ different from this RGBCW bulb - https://www.action.com/nl-nl/p/lsc-smart-connect-slimme-multicolor-ledlamp-1b6ebaed/ . That's LSC/2578539-970719, and [its schema is in this file](https://github.com/khalednassar/tuya-cloudcutter/blob/808634da2cb994feb4b030f5de565763c597faf0/device-profiles/LSC/2578539-970719/tuya.device.active.json).

Would be nice to confirm if this currently works w/ the stock firmware and the device locally configured.

tony-fav, 2022-04-04:
Thanks!

[dump_extract_dir.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8410165/dump_extract_dir.zip)

Better dump using your tools 🤦‍♂️

I've got a new bulb on the way to test with. This is a dump from a transplanted/dissected/destroyed bulb :-)

khalednassar, 2022-04-04:
Should be good to go now!
Interesting to note, the DPID schema was pretty similar to the LSC one. This STITCH bulb had just one extra dpid (27) which the LSC one didn't have, and one less dpid (41) that the LSC one had. The rest were identical.

tony-fav, 2022-04-10:
Everything seems to work. I ... uh ... don't actually have the chip in a bulb to test but it cut, connected to my wifi, and I can play with the datapoints in tinytuya.

[image]

closed 2022-04-10 by tony-fav