tuya-cloudcutter/tuya-cloudcutter #11: FEIT OM60 A19 RGBW - WB2L based bulb

kwiksilver, 2022-04-04:
WB2L: https://developer.tuya.com/en/docs/iot/wb2l-datasheet?id=K9duegc9bualu

FEIT OM60 A19 RGBW Bulb: https://www.feit.com/product/smart-wifi-bulb-color-changing-tunable-white-2/

Firmware Dump: [Feit_OM60_RGBW_SYW-A19RGBWAGT2_Dump1.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8410483/Feit_OM60_RGBW_SYW-A19RGBWAGT2_Dump1.zip)

tfrew-r7, 2022-04-04:
Would be good to test the offsets for this against the other feit bulbs lowes sells

kwiksilver, 2022-04-04:
Agreed. This bulb came directly from Feit as an RMA for an ESP based bulb so naturally it sat on my shelf untouched until this exploit came out.

Historically when there were a lot of ESP based Feit bulbs around, the costco, lowes, and menards ones all had slightly different templates for tasmota. It would be interesting to see if that's the case in these bulbs.

khalednassar, 2022-04-04:
Should be good to go, full profile added.

tony-fav, 2022-04-10:
Worked on my Feit bulb of the same model number from Menards.

kwiksilver, 2022-04-10:
Working here on my device as well. Closing

closed 2022-04-10 by kwiksilver