tuya-cloudcutter/tuya-cloudcutter #17: Instructions for adding new devices ?

AndyRPH, 2022-04-05:
Are there instructions for pulling a firmware dump without opening a device to be able to contribute towards the expansion of supported devices? 

I've got a handful of tuya devices that don't support the conversion to tasmota, hoping that's because they're using chips that this project might support.  But with tuya having already patched the exploit, time may be ticking to stock up on some devices that this will work on.  

(Specifically I'm keen on normal toggle looking dimmer switches that are wifi, not zigbee or zwave, like this: 3-Way Smart WiFi Toggle Dimmer Light Switch, in-Wall, No Hub Required, Compatible with Alexa/Google Home, ETL and FCC Listed (WF31TS) https://www.amazon.com/dp/B08T1JBD9P/ref=cm_sw_r_cp_api_i_91SR38557T9HJPC4VSJK?_encoding=UTF8&psc=1)  

I can't find my notes from when I took it apart but I believe it was a tuya WB2S chip, which should be compatible if I'm understand this project right!

digiblur, 2022-04-05:
Those are Beken chips so yes they would be for this project. You would need to solder to the chip and dump the firmware.

AndyRPH, 2022-04-05:
Ah rats, hoping there was an OTA firmware dump via this exploit.

digiblur, 2022-04-05:
> Ah rats, hoping there was an OTA firmware dump via this exploit.

at this point you'll have to wait for someone else to dump that switch then.

tjclement, 2022-04-08:
Note that if you just want to find out if the device is vulnerable so you can stock up on a bunch of them, you can check that wirelessly: https://github.com/khalednassar/tuya-cloudcutter#testing-if-a-device-is-exploitable

AndyRPH, 2022-04-08:
Good to know!  Is it not possible to get a firmware dump using the exploit, or just not a priority to write a dump tool that uses the exploit?

tony-fav, 2022-04-08:
The exploit depends upon the firmware dump, not the other way around as you suggest.

AndyRPH, 2022-04-09:
Alright you've motivated me, I may give a dumping the firmware a shot this weekend.

AndyRPH, 2022-04-09:
I've got it apart and soldered leads to the WB2S pins needed.  How do I build the firmware dump tool for terminal use on osx?  I've got an ft232 USB adapter I've used for tasmota on other things.

khalednassar, 2022-04-12:
For most devices it's possible, there's might be some interference from other components but usually it works. What you can do is run the tool, and connect the CEN pin to GND briefly (tap them together), which should get it running. Otherwise, just power off the module on and off a few times while the tool is running, but don't power off your FT232.

mattvirus, 2022-04-12:
also for some chips i've had success by getting everything connected, running the dump tool, and then disconnecting/reconnecting VCC from the programmer pin.  Forces the chip to reset without resetting the programmer itself, and the dumping software (that is already running) is able to interface with the beken chip as it resets.

btsimonh, 2022-04-12:
yep, they can be a right royal pain.  Triple check everything and try again.  also can be worth trying a different adaptor. (I have one which is much less reliable than my others).

kwiksilver, 2022-04-12:
I've had success dumping with FT232RL and CP2012 based dongles. The 340C based ones were not working for me at all. It may come down to timing but the FT232RL worked every time for me.

AndyRPH, 2022-04-12:
Yeah, I'm keeping the FT232 plugged in and just disconnecting and reconnecting the VCC pin.

btsimonh, 2022-04-12:
If you have a second adaptor, and access to the second serial port (tx only) , it's worth looking at the boot log.

Cossid, 2022-12-01:
Closing stale ticket, instructions have changed drastically since this ticket was opened.

closed 2022-12-01 by Cossid