tuya-cloudcutter/tuya-cloudcutter #21: WB2L dump from Tuya A-60 RGB bulb

ferbulous, 2022-04-06:
WB2L: https://developer.tuya.com/en/docs/iot/wb8p-module-datasheet?id=K9fwx4f89tvzd

Tuya E27 A-60 Bulb: https://www.lazada.com.my/products/i2537570470-s11249633916.html

Used https://github.com/khalednassar/bk7231tools to dump default size (python bk7231tools.py read_flash -d COM7 -s 0 -c 512 wb2l.bin

[wb2l.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8424648/wb2l.zip)

khalednassar, 2022-04-06:
Just added the profile, which interestingly seems to match the LSC 970715 device. Maybe that's a generic app image, would be interesting to see if more devices match.

@ferbulous could you validate if that profile under `Tuya-Generic` brand works?

ferbulous, 2022-04-07:
Hi, i'm still trying to get it to work, I'm just having this error on the last part 

```
`[?] Select the article number of your device: E27-A60-RGBCWW-V1
 > E27-A60-RGBCWW-V1

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in b                                                                             etween, to enable AP mode. Repeat if your device's SSID doesn't show up within ~                                                                             30 seconds.
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Found access point name: "A-1FCD", trying to connect..
Device 'wlp2s0b1' successfully activated with '004b119d-410e-4c7c-8916-45179fb99                                                                             691'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/zJ32k48A7UNyMXoM1Ei3.deviceconfig
Saved device config in /work/configured-devices/zJ32k48A7UNyMXoM1Ei3.deviceconfi                                                                             g
==> Turn the device off and on again once. Repeat 6 more times if your device's                                                                              SSID doesn't show up within ~5 seconds.
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Found access point name: "A-1FCD", trying to connect..
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Scanning for "(A-)|(SmartLife-)|(iHome-)" SSID...
Found access point name: "A-1FCD", trying to connect..
Device 'wlp2s0b1' successfully activated with '004b119d-410e-4c7c-8916-45179fb99                                                                             691'.
Connected to access point.
payload='{"ssid": "cloudcutter-flash", "passwd": "", "token": "AAAAAAAA"}'
Configured device to connect to 'cloudcutter-flash'
Device is connecting to 'cloudcutter-flash' access point
Cutting device off from cloud..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutter-flash'. Thi                                                                             s script will then show the activation requests sent by the device, and tell you                                                                              whether local activation was successful.
Using WLAN adapter: wlp2s0b1
Configuration file: /dev/stdin
Using interface wlp2s0b1 with hwaddr 88:53:95:28:92:7d and ssid "cloudcutter-fla                                                                             sh"
wlp2s0b1: interface state UNINITIALIZED->ENABLED
wlp2s0b1: AP-ENABLED
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/src/cloudcutter/__main__.py", line 270, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 147, in __configure_local_device_or_                                                                             update_firmware
    http_server.listen(80)
  File "/root/.local/share/virtualenvs/src-NVTF7jWz/lib/python3.9/site-packages/                                                                             tornado/tcpserver.py", line 151, in listen
    sockets = bind_sockets(port, address=address)
  File "/root/.local/share/virtualenvs/src-NVTF7jWz/lib/python3.9/site-packages/                                                                             tornado/netutil.py", line 161, in bind_sockets
    sock.bind(sockaddr)
OSError: [Errno 98] Address already in use
Oh no, something went wrong with detaching from the cloud! Try again I guess..
`
```

khalednassar, 2022-04-07:
It looks like you have something already listening on your system on port 80, maybe some http server?
Make sure you have nothing listening on ports 80, 443 and 1883 before running the container.

ferbulous, 2022-04-11:
Profile is confirmed working using the tool

closed 2022-04-11 by ferbulous