tuya-cloudcutter/tuya-cloudcutter #27: WB3S - Treatlife ST10 Lightstrip Dump

mattvirus, 2022-04-08:
WB3S:  https://developer.tuya.com/en/docs/iot/wb3s-module-datasheet?id=K9dx20n6hz5n4

Treatlife ST10 Lightstrip:
https://www.treatlife.tech/products/treatlife-smart-led-strip-lights-16-4ft-color-changing-wifi-rgb-led-lights-tv-backlight-compatible-with-alexa-google-home-for-voice-control-for-bedroom-kitchen-party

https://www.amazon.com/Treatlife-Changing-Backlight-Control-Bedroom/dp/B08HGQ7H8F

Dumped with bk7231tools
[treatlife_st10.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8455746/treatlife_st10.zip)

digiblur, 2022-04-09:
Same device model but secondary if needed
[tlife_ledstrip.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8457708/tlife_ledstrip.zip)

khalednassar, 2022-04-12:
@digiblur @mattvirus could you please verify that https://github.com/tuya-cloudcutter/tuya-cloudcutter/pull/51 works?

digiblur, 2022-04-12:
Will do now that I have a working Pi3 with this.

MoogleTroupe, 2022-04-20:
bump, did this profile cause the device to show an A- ap after the exploit ran?

closed 2022-11-30 by Cossid