tuya-cloudcutter/tuya-cloudcutter #30: Actual firmware change

btsimonh, 2022-04-09:
Has there been any progress on flashing custom firmware into exploited devices?
If I knew where to start, I'd suggest writing it as a Node-Red flow?
s

khalednassar, 2022-04-09:
Flashing custom firmware already works, just the instructions are missing. Will add that soon (a PR is most welcome too!)

But really, it's not very different from the `run_detach` flow.

All you have to do is compile your firmware and copy the UG file to the `custom_firmware` directory, then use the `run_flash.sh [optional wifi interface name]` script which should take you through the prompts to select a profile for the device then the desired firmware file to OTA.

btsimonh, 2022-04-09:
ref NR, I'm very comfortable with hacking around in that, and much less with Python :).

Ref instructions, maybe enable the wiki here, and make it publicly editable, then we may get some people contributing docs which could later become more formal?

Also a good place to document devices & experiences?

khalednassar, 2022-04-09:
Good catch and agreed re: an open wiki. I've added the default homepage for now, pending these instructions and some reorganization :)

tjclement, 2022-04-10:
Huh, odd, I thought I added instructions for custom flashing when we implemented it a week ago. Thanks for the quick recap @khalednassar

ferbulous, 2022-04-10:
> Flashing custom firmware already works, just the instructions are missing. Will add that soon (a PR is most welcome too!)
>
> But really, it's not very different from the `run_detach` flow.
>
> All you [...]

HI, trying out custom firmware flashing with run_flash script but it looks like it got stuck or just not giving any update after some time

Tested on raspian-buster, UA firmware (v1.0.4), device= A-60 (generic tuya)
Thoughts?

https://pastebin.com/j7wapzCn

khalednassar, 2022-04-11:
> HI, trying out custom firmware flashing with run_flash script but it looks like it got stuck or just not giving any update after some time
>
> Tested on raspian-buster, UA firmware (v1.0.4), device= [...]

@ferbulous You're actually very close! You need to use the OpenBK7231T_UG_1.0.4.bin file instead of the UA file and that's hopefully it.

Why? The UA file is formatted for direct flashing via serial or SPI, so it has the so-called RBL header at the end and is littered with CRC16 checksums every 32-byte blocks. Meanwhile the UG file is a tuya-wrapped RBL file, which the stock firmware unpacks after stripping the tuya header and the beken bootloader will load after a reset.

pethans, 2022-04-28:
I grabbed tasmota.bin out of tuya-convert and `run_detach` informed me it is not a valid UG file so it must have been using another process, or converting on the fly? I have almost 10 TYWE3S TreatLife devices I'm waiting to flash with Tasmota and would be more than happy to test this capability, but I'm not sure of a way to obtain/convert/compile a UG binary.

MoogleTroupe, 2022-04-29:
tasmota is for esp8266/esp32 chipset devices, not beken bk7231 chipset devices. The only third party open source firmware available for the beken chips right now is openshw https://github.com/openshwprojects/OpenBK7231T_App

pethans, 2022-04-29:
Thanks — yes, TYWE3S is esp8266-based and my DS01c switches are confirmed to have this chip, but they are notoriously hard to flash because the tx/rx pins are used for Tuya MCU (I've seen the suggestions to use an external power supply, but thought this method might be a worth trying first). I only realized after my comment that there's no 'dump' yet for this dimmer. I understand if the focus here is bk7231 chips, but there may be a use case for tasmota as well.

tony-fav, 2022-04-29:
I think you are not understanding. This *exploit* ONLY works on BK7231 chips. It doesn't matter if you have a firmware dump from the ESP, you would not be able to use this exploit.

pethans, 2022-04-29:
You're right, I didn't see that explicitly noted anywhere in the documentation. Thanks for clarifying.

MoogleTroupe, 2022-04-29:
that ds01c should be super easy to flash, just ground the mcu (as in "nvreset->gnd") so it gets stuck in a boot loop then you can talk rx/tx to the esp. i recommend reading a tutorial like https://www.digiblur.com/2020/07/the-tasmota-fan-controller-ive-been.html - a ds03, but probably similar board design.

Gamerayers, 2022-09-06:
I was looking for the info on how to flash, and still didn't see it.  I have a BUNCH of SL10 bulbs which I've love to toss the custom firmware on them.
I saw this: https://newreleases.io/project/github/openshwprojects/OpenBK7231T_App/release/1.12.34 has the custom firmware, but haven't seen the instructions.  Also, not entirely sure which version to use for a WB9P chip.
I've tried to desolder method, and really was unable to do anything with the it.  Sadly.  Even though I've done similar with ESP chips.

tjclement, 2022-11-21:
Brief instructions on how to flash custom firmware were added a while back: https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md#flashing-custom-firmware, so I'm closing this issue. If anything's unclear, please reach out in a new issue. Thanks!

closed 2022-11-21 by tjclement