tuya-cloudcutter/tuya-cloudcutter #31: Avatar ASL04 TV Backlight dump

btsimonh, 2022-04-09:
Zip containing original firmware dump plus images and some text details:
[AvatarASL04.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8456601/AvatarASL04.zip)

Are there instructions for analysing a dump ourselves?
e.g. hex strings to search for the two required jump points?

khalednassar, 2022-04-09:
RE instructions: it's a bit of a process, so making a helpful guide will take some time. That being said, there's some attempt at automating it https://github.com/khalednassar/cloudcutter-bk7231-haxomatic , which doesn't work for all app code versions we've seen yet but the code describes somewhat of a process.

These dumps all don't contain the config section (0x1EF000+), which is fine for making the payload for the exploit, but the profile still needs a schema for stock firmware usage locally - [explanation and some caveats here](https://github.com/khalednassar/tuya-cloudcutter#device-dumps). They can be guessed but it's easier done through the API. You can pull the schema yourself with the device's UUID + authkey if sharing the full dump is not an option with this script: https://github.com/khalednassar/tuya-cloudcutter/blob/main/profile-building/pull_active_response.py

khalednassar, 2022-04-09:
Nevermind the config section part above, I see that this dump does include part of the config section which seems to be just enough!

btsimonh, 2022-04-09:
Are the hex codes for the instructions we need always the same?
Seems to me a simple search for a specific byte sequence should be enough, or am I missing something....

khalednassar, 2022-04-09:
Yes and no. There are some patterns to the instruction sequences but a specific byte sequence match won't suffice. There's some variance to the registers used, varied usage of the stack in the target sequences and instruction shifting. Of course, you could still do iterative fuzzy matches on a bunch of byte sequences but it isn't simpler to do compared to a higher level matching on instruction patterns.

The automation tool linked above does that exactly that, fuzzy matching on instruction patterns which essentially boils down to fuzzy matching on byte sequences but has the nice property of being more understandable.

khalednassar, 2022-04-12:
@btsimonh could you please check if https://github.com/tuya-cloudcutter/tuya-cloudcutter/pull/52 works?

khalednassar, 2022-04-17:
Working per convo on discord :)

closed 2022-04-17 by khalednassar