tuya-cloudcutter/tuya-cloudcutter #32: Calex UK power strip (Tescos) dump+images

btsimonh, 2022-04-09:
firmware dump plus images & bootlogs:
[Calex-4way-socket-usb.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8456604/Calex-4way-socket-usb.zip)

khalednassar, 2022-04-11:
@btsimonh Profile added to PR https://github.com/tuya-cloudcutter/tuya-cloudcutter/pull/45 - can you give it a try and confirm if it works?

MoogleTroupe, 2022-04-20:
bump, did this profile cause the device to show an A- ap after the exploit ran?

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid