tuya-cloudcutter/tuya-cloudcutter #33: Calex RGBWW LED strip (tesco uk)

btsimonh, 2022-04-09:
Firmware dump only:
[CalexOriginalLEDStripFirmware.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8456611/CalexOriginalLEDStripFirmware.zip)

MoogleTroupe, 2022-04-20:
bump, did this profile cause the device to show an A- ap after the exploit ran?

btsimonh, 2022-10-09:
Bought another of these yesterday.
(successful) Exploit log below.

device appeared as 'wlan0' in my router.

Will now try to OpenBeken it....
@notkmhn - Once I have openbeken on, I may be able to pull the config section if it still exists; although it's never seen the  app or the cloud?  What flash address is the config?

<details>
  <summary>Click to expand log</summary>
```
Successfully built docker image
[?] Select the brand of your device: Calex
   Arlec
   Avatar
   Ayococr
 > Calex
   CASENL
   CLOUDY_BAY
   DIY
   Euarne
   Fcmila
   Feit
   FLSNT
   Gosund
   Helloify

[?] Select the article number of your device: Holland-LED-Strip-900023.1
   E27-A60-RGBCCT-429004.1-v2.0
 > Holland-LED-Strip-900023.1
   UK-900018.1-v1.0

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Wiping NetworkManager configs
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-BD20", trying to connect..
Device 'wlan0' successfully activated with '4e905258-7741-4d01-bd1e-011717604924'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/reDRgL8mdMWJ.deviceconfig
Saved device config in /work/configured-devices/reDRgL8mdMWJ.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds. For smart plugs long press the reset button on the device for about 5 seconds. See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
Wiping NetworkManager configs
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-BD20", trying to connect..
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
Wiping NetworkManager configs
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "A-BD20", trying to connect..
Device 'wlan0' successfully activated with '742b3353-c388-4d8b-9644-526408fa4fee'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Cutting device off from cloud..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the activation requests sent by the device, and tell you whether local activation was successful.
Using WLAN adapter: wlan0
Oct  9 07:56:21 dnsmasq[15]: started, version 2.80 cachesize 150
Oct  9 07:56:21 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Oct  9 07:56:21 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Oct  9 07:56:21 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Oct  9 07:56:21 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
[... 102 lines cut ...]
Oct  9 07:56:48 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
[I 221009 07:56:48 web:2239] 200 POST /v1/url_config (10.42.42.24) 5.95ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.active&et=1&t=81&uuid=reDRgL8mdMWJ&v=4.4&sign=73d250a5c6fda45e8af6d97d255f0a1d
[LOG] ==== Client request body ===
{'token': 'AAAAAA', 'softVer': '1.1.6', 'productKey': 'keyqwsdet7drnvue', 'protocolVer': '2.2', 'baselineVer': '40.00', 'productKeyStr': 'keyqwsdet7drnvue', 'hid': '508a067bbd20', 'modules': '[{"type":9,"softVer":"1.1.6","online":true}]', 'devAttribute': 515, 'cadVer': '1.0.2', 'cdVer': '1.0.0', 'options': '{"isFK":true}', 't': 81}
[LOG] ==== End client request body ===
tuya.device.active - response: {'result': {'schema': '[{"type":"obj","mode":"rw","property":{"type":"bool"},"id":20},{"mode":"rw","property":{"range":["white","colour","scene","music"],"type":"enum"},"id":21,"type":"obj"},{"mode":"rw","property":{"type":"string","maxlen":255},"id":24,"type":"obj"},{"mode":"rw","property":{"type":"string","maxlen":255},"id":25,"type":"obj"},{"type":"obj","mode":"rw","property":{"m [... cut ...]
[LOG] ==== Client request body ===
{'t': 1665302211}
[LOG] ==== End client request body ===
tuya.device.uuid.pskkey.get - response: {'result': {'pskKey': ''}, 't': 1665302212, 'success': True}
[I 221009 07:56:52 web:2239] 200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=1665302211&uuid=reDRgL8mdMWJ&v=1.0&sign=5eb4b39a2541682e0f5cfb3c8578310a (10.42.42.24) 9.77ms
device_id='FxGRwY7BLeqOFfVABzXy', ip='10.42.42.24', local_key='2EiJ28ReFLQr16xV'
Device should be successfully onboarded on WiFi AP!
```
</details>

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid

spanner3003, 2024-05-17:
I'm not sure where to put these file that a pulled off the Calex smart multi color LED strip I brought from tesco years ago but i used esphome-kickstarter and UPK2ESPHome grab for kickstarter option hope they are helpful.
[Merkury EW003 LED Strip - schema ID 000002hqnb.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15355599/Merkury.EW003.LED.Strip.-.schema.ID.000002hqnb.json)
[Merkury EW003 Smart LED Strip Storage Data.json](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/15355600/Merkury.EW003.Smart.LED.Strip.Storage.Data.json)
This is the box they came in:
[image]