tuya-cloudcutter/tuya-cloudcutter #36: WB3S - Martin Jerry Fan Controller US-FC-01

digiblur, 2022-04-09:
https://www.amazon.com/Ceiling-Control-Martin-Jerry-SmartLife/dp/B09BDRTZK2
Firmware  
[mj_fan_us_fc_01.zip](https://github.com/khalednassar/tuya-cloudcutter/files/8457766/mj_fan_us_fc_01.zip)

MoogleTroupe, 2022-04-10:
can you confirm https://github.com/khalednassar/tuya-cloudcutter/pull/40 gets you A- after step one

MoogleTroupe, 2022-04-20:
bump, did this profile cause the device to show an A- ap after the exploit ran?

digiblur, 2022-04-21:
Don't have a way to test it, tried to get the chip into pairing mode on the burner but wasn't able to.

MoogleTroupe, 2022-04-21:
that's okay, it has the same APP bin as the [DS02F/profile](https://github.com/tuya-cloudcutter/tuya-cloudcutter/pull/40/files#diff-3c433c1380cfe9b8b623b305b5bc0fc41d6f4b73b041f997625f89c51947a534) - only need one of these to be validated.

digiblur, 2022-04-21:
no A- access point, tried several times and keeps coming back to SmartLife-xxxx

MoogleTroupe, 2022-04-21:
check against https://github.com/tuya-cloudcutter/tuya-cloudcutter/pull/71 now

closed 2022-11-30 by Cossid