tuya-cloudcutter/tuya-cloudcutter #55: CB3S - Lonsonho Tuya Smart WiFi Switch 4 Gang dump

ferbulous, 2022-04-12:
CB3S: https://developer.tuya.com/en/docs/iot/cb3s?id=Kai94mec0s076

Lonsonho Tuya Smart WiFi Switch 4 Gang EU: https://www.aliexpress.com/item/1005003590644405.html

Used bk7231tools to dump:
>python bk7231tools.py read_flash -d COM7 -s 0 -c 512 cb3s.bin --no-verify-checksu
[cb3s.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8474833/cb3s.zip)
m

khalednassar, 2022-04-12:
Added the profile, could you please validate that it works?
Also, are you sure this is a CB3S? Device config section states it's a WB3S. Not that it matters much, but just to confirm.

ferbulous, 2022-04-12:
My apologies, the CB3S is from another device that i haven’t uploaded yet.
This is indeed WB3S

MoogleTroupe, 2022-04-20:
bump, did this profile cause the device to show an A- ap after the exploit ran?

closed 2022-11-30 by Cossid