tuya-cloudcutter/tuya-cloudcutter #58: CBU - Generic Tuya Smart WiFi Switch 4 Gang dump

ferbulous, 2022-04-13:
CBU: https://developer.tuya.com/en/docs/iot/cbu-module-datasheet?id=Ka07pykl5dk4u

Generic Tuya Smart WiFi Switch 4 Gang EU: https://www.alibaba.com/product-detail/Factory-Tuya-Wifi-4-Gang-Touch_1600225541456.html

Used bk7231tools to dump:
python bk7231tools.py read_flash -d COM7 -s 0 -c 512 cbu_4gang.bin --no-verify-checksum
[cbu_4gang.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8482511/cbu_4gang.zip)

khalednassar, 2022-04-17:
Seems like a specific brand, so the profile is created under CASENL instead of generic. Please check if it works.

MoogleTroupe, 2022-04-20:
bump, did this profile cause the device to show an A- ap after the exploit ran?

ferbulous, 2022-05-03:
> bump, did this profile cause the device to show an A- ap after the exploit ran?

Yep

closed 2022-05-03 by ferbulous