tuya-cloudcutter/tuya-cloudcutter #59: CB3S - Generic Tuya Smart WiFi Switch 3 Gang dump

ferbulous, 2022-04-13:
CB3S: https://developer.tuya.com/en/docs/iot/cb3s?id=Kai94mec0s076

Generic Tuya Smart WiFi Switch 3 Gang EU: https://www.aliexpress.com/item/1005003379877235.html

Used bk7231tools to dump:
python bk7231tools.py read_flash -d COM7 -s 0 -c 512 cb3s_3gang_switch.bin --no-verify-checksum
[cb3s_3gang_switch.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8482527/cb3s_3gang_switch.zip)

khalednassar, 2022-04-17:
Profile created under Tuya Generic, please verify that it works.

MoogleTroupe, 2022-04-20:
bump, did this profile cause the device to show an A- ap after the exploit ran?

Cossid, 2022-11-22:
No user verification, but we have a full dump with schema, it will be regenerated for universal profiles.  Closing this issue.

closed 2022-11-22 by Cossid

blakadder, 2022-11-22:
The model is LSPS5

Cossid, 2022-11-22:
Noted, will include that in the model name for universal.

divadiow, 2024-11-28:
> CB3S: https://developer.tuya.com/en/docs/iot/cb3s?id=Kai94mec0s076
> 
> Generic Tuya Smart WiFi Switch 3 Gang EU: https://www.aliexpress.com/item/1005003379877235.html
> 
> Used bk7231tools to dump: python bk72
[ltchiptool_bk72xx_2024-11-28_21-39-35.zip](https://github.com/user-attachments/files/17952415/ltchiptool_bk72xx_2024-11-28_21-39-35.zip)
31tools.py read_flash -d COM7 -s 0 -c 512 cb3s_3gang_switch.bin --no-verify-checksum [cb3s_3gang_switch.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8482527/cb3s_3gang_switch.zip)

this firmware offers an update to v1.1.0 from v1.0.5
[image]

Cossid, 2024-11-29:
Split into `v1.0.5` and `v1.1.0`