tuya-cloudcutter/tuya-cloudcutter #61: TH06 clock calendar humidity temperature sensor LCD [WB3S]

openshwprojects, 2022-04-16:
This device:

https://www.elektroda.pl/rtvforum/topic3819498.htm

[TH06_bk7231s_dump-2022- 3- 2-23-32-42.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8500132/TH06_bk7231s_dump-2022-.3-.2-23-32-42.zip)
l

Please check if the flash is read correctly, I did it a month ago or so, so I am not sure

khalednassar, 2022-04-17:
Flash seems complete off indeed, do you happen to have another dump? Otherwise I'm afraid it's not possible to build a payload for this one

openshwprojects, 2022-04-18:
No, it was flashed by OpenBK long time ago.

What's wrong with the flash? Are you unable to decrypt that or what?

khalednassar, 2022-04-18:
The app code decrypts incorrectly and seems to have a decent amount of errors, even as early on as the first ~300ish bytes based on the errors I got. I'm not sure exactly what's wrong since there's no RBL header to check CRC against either.

closed 2022-12-01 by Cossid