tuya-cloudcutter/tuya-cloudcutter #62: Qiachip CB2S version (there are differenet version) just a simple relay BK7231N

openshwprojects, 2022-04-16:
this device: https://www.elektroda.pl/rtvforum/viewtopic.php?t=3874289&highlight=

[qiachip_cb2s_beforePairingIthinkSo.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8500137/qiachip_cb2s_beforePairingIthinkSo.zip)

khalednassar, 2022-04-17:
Profile added! Is it possible for you to test it on the stock firmware on your device?

MoogleTroupe, 2022-04-20:
bump, did this profile cause the device to show an A- ap after the exploit ran?

ElGranPepo, 2022-09-08:
it looks like there is a dump of this device and a pinout in https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/128

if we are able to exploit it, is there a way to upload a firmware to it like https://github.com/kuba2k2/libretuya  OTA?

Thank you for all your help.

Cossid, 2022-11-30:
An update has been pushed that should include a profile for this device.  Please update your repo and give it a try.  Please see the instructions page ( https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md ) for any command or input changes.

If the new profile does not work, please re-open this ticket with details (including logs) of the failure.
Any feedback is appreciated.

closed 2022-11-30 by Cossid

jkwim, 2022-12-16:
Can you correct the naming please?
```
 > QIACHIP
   Qiachip
```
There are two entries with same name

Cossid, 2022-12-16:
Done.

jkwim, 2022-12-16:
> Done.

Wow! So quick!
Thank you!

I was running the script for the second time and was wondering why only a single entry now :-)