tuya-cloudcutter/tuya-cloudcutter #63: BK7231N Airam Smart A60 827-865 RGB 806lm E27 OP

faster-harder-coder, 2022-04-17:
See https://www.elektroda.com/rtvforum/viewtopic.php?p=19977811

At first boot it is blinking rapidly and only showing a bluetooth device named "TY". After being switched on/off three times it shows a Smartlife-xxxx Wifi AP. Tested "test_device_exploitable.py" which indeed stops the blinking and starts again after ~60s. Also tested the Tuya-Generic / E27-A60-RGBWW-V1 profile, but it didn't work.

Dumped with "OpenBekenIOT/hid_download_py/uartprogram"
uartprogram -r -s 0 -d /dev/ttyUSB0 -l1ef000 -u dump3.bin

[dump3.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8501410/dump3.zip)

khalednassar, 2022-04-29:
@faster-harder-coder is it possible to provide a dump with the config section after 0x1ef000? If so please keep in mind (per the README) that it might have your SSID and password if you've already configured the device on tuya cloud.
It's necessary to build a proper schema for the device to work with stock firmware. If not, can also use a schema from a similar device but that might or might not work well.

Cossid, 2022-12-10:
This post has been tagged as new-dump-needed and has had no response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-10 by Cossid