tuya-cloudcutter/tuya-cloudcutter #64: Dumping WB8P from Energizer EAW2-1001-SWT Bulb - Running into issues

TheN00bBuilder, 2022-04-17:
Hey all,

I'm attempting to dump this light bulb for not only this, but another project I'm working on. I've gotten TX1 and RX1 soldered up along with GND and VCC (going to 3.3V) but cannot get this bulb to connect via using the BKWriter tool or the hid_download_py program. 

I've tried both at all possible speeds, flipping around TX and RX, but nothing has worked. I always get a fail to init on the BKWriter, or "Cannot Get Bus" from the hid_download_py tool. I've tried rebooting during these processes when it is trying to connect, and nothing has worked so far.

My options for the command line are `python3 uartprogram -d COM4 -s 0x0 -u -r bindump.bin` along with various speeds. All of them have the same issue.

The lightbulb still responds within the app, so I know the chip is not dead.

Any ideas or suggestions?

ferbulous, 2022-04-18:
I could never get bkwriter to work,  use https://github.com/khalednassar/bk7231tools instead
once it's powered up, use this command

python bk7231tools.py read_flash -d COM7 -s 0 -c 512 dump.bin

if your device is bk7231n, use this

python bk7231tools.py read_flash -d COM7 -s 0 -c 512 dump.bin --no-verify-checksum f

openshwprojects, 2022-04-18:
Have you tried CEN method?

TheN00bBuilder, 2022-04-18:
Thanks everyone for your comments! @ferbulous thanks for the exact commands to run, sadly still no luck using bk7231tools, I always get a TimeoutError. I have not tried the CEN method @openshwprojects . Do you have a guide on how to do that?

I'm beginning to think they did the "responsible" thing and blew the e-fuse, but I have no clue if these chips have those.

TheN00bBuilder, 2022-04-18:
> These chips don't have something like GPIO0 on an ESP8266 to put them into "bootloader mode." What they do have is a really really small window on boot where you can send the right packets to be able to load or dump firmware. That's what bk7231tools does. Try the following process:
> 
> * Connect 3v3 and GND to the chip. Connect TX on USB adapter to RX1 on chip. Connect RX on USB adapter to TX1 on chip.
> * Start bk7231tools
> * Take another lead connected to GND and tap the CEN pin
> 
> What this does is have bk7231tools sitting and waiting to talk to the chip as fast as possible then you restart the chip by grounding the enable pin momentarily.

Perfect! Just tried this and got the full chip to dump. Thank you so much for the guide!! 

I realized that my WiFi password and network are in plaintext on the .BIN, would it be okay for me to block them out with a's before I upload the dump here?

tony-fav, 2022-04-18:
https://github.com/khalednassar/bk7231tools#dissecting-flash-dumps <-- do the dissection and check if the resulting files have your info. If not, upload them instead. A dump that has been connected to the cloud is best because it's less work for @khalednassar to get the DPID table needed for the device to work locally.

TheN00bBuilder, 2022-04-18:
> https://github.com/khalednassar/bk7231tools#dissecting-flash-dumps <-- do the dissection and check if the resulting files have your info. If not, upload them instead. A dump that has been connected [...]

Got it. Sorry, should have read more into that link. Those files did not have the important info in them, so I zipped them up and here they are. [dump_extract_Energizer_EAW2-1001-SWT.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8506825/dump_extract_Energizer_EAW2-1001-SWT.zip)

MoogleTroupe, 2022-04-19:
@TheN00bBuilder can you also upload the full dump with the password changed to a same length value

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid