tuya-cloudcutter/tuya-cloudcutter #68: WB3L - Arlec Grid Connect Smart 9W CCT LED Downlight

talltechdude, 2022-04-19:
Arlec Grid Connect Smart 9W CCT LED Downlight

https://www.bunnings.com.au/arlec-grid-connect-smart-9w-cct-led-downlight_p0168694

[ArlecCCT_Downlight.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8512452/ArlecCCT_Downlight.zip)

Also uses `GRID-` as a SSID preamble (presumably short for Grid Connect) so that needs to be added to [ap_preambles.txt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/ap_preambles.txt).

MoogleTroupe, 2022-04-19:
check if #69 turns the ap into A-

talltechdude, 2022-04-19:
Yep, thanks. #69 now gets an `A-` and the device connects to the `cloudcutterflash` AP but the tool seems to hang after the first communication:
```
Using PSK v1 - Received PSK ID version 01
[LOG] Client request PATH: /d.json - QS: a=tuya.device.active&et=1&t=67&uuid=GniDm2G0qKcl&v=4.4&sign=1d926b6d4fedb81ad550cc2bafe53df3
[LOG] ==== Client request body ===
{'token': 'AAAAAA', 'softVer': '2.9.15', 'productKey': 'keytg5kq8gvkv9dh', 'protocolVer': '2.2', 'baselineVer': '40.00', 'productKeyStr': 'keytg5kq8gvkv9dh', 'hid': '508a06134796', 'modules': '[{"type":9,"softVer":"2.9.15","online":true}]', 'devAttribute': 515, 'cadVer': '1.0.2', 'cdVer': '1.0.0', 'options': '{"isFK":true}', 't': 67}
[LOG] ==== End client request body ===
tuya.device.active - response: {'success': True, 't': 1650376438}
```

Tried on multiple devices, stuck at same point

MoogleTroupe, 2022-04-19:
can you dm me a dump of one that has been connected to tuya-cloud once before, i will extract the json schema needed for the second step to work from that

talltechdude, 2022-04-19:
> can you dm me a dump of one that has been connected to tuya-cloud once before

DM sent to the email on your GitHub profile

MoogleTroupe, 2022-04-19:
pull latest, should work now

talltechdude, 2022-04-19:
Confirmed works now in both `run_detach` and `run_flash` to convert to [OpenBK7231T_App](https://github.com/openshwprojects/OpenBK7231T_App)

closed 2022-04-19 by talltechdude