tuya-cloudcutter/tuya-cloudcutter #72: WB8P - LEDVANCE Smart+ WW/CW Classic A60 B27 Tunable White Light Bulb

ihrapsa, 2022-04-22:
I've attached the bin dump of one of these WB8P bulbs.
[ledvance_wwcw.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8537393/ledvance_wwcw.bin.zip)

Dumped with [bk7231tools](https://github.com/khalednassar/bk7231tools)
`python bk7231tools.py read_flash -d /dev/tty.usbserial-0001 -s 0 -c 512 dump.bin`

This is how it started:
```
Connected! Chip info: BK7231S_1.0.5
Flash read start address 0x0 is not greater than 0x10000 - adding 0x2000000 to bypass bootloader checks
Reading 4k page at 0X2000000 (0.00%)
Reading 4k page at 0X2001000 (0.20%)
Reading 4k page at 0X2002000 (0.39%)
....
```

PS. I tested it with "test_device_exploitable.py" but it did not stop the flash.

Thanks!

khalednassar, 2022-04-23:
App RBL is due to upgrading as hypothesized indeed. New updates to bk7231tools should fix the issue with dissection for this one.

khalednassar, 2022-04-23:
Profile has been added to main, please check if it works for you.

ihrapsa, 2022-04-23:
> Profile has been added to main, please check if it works for you.

It works!! the upgraded bulb is down to earth now! 
Thank you!!! 

This weekend I don't have access to the bulbs that have not been upgraded so I can't test on them yet. I wonder if this profile would work on the new 'virgin' bulbs or I would have to provide a new bin dump.

Also, I'd like to learn the magic behind profile building for future PRs.

Really appreciate your work!!!

khalednassar, 2022-04-23:
Thank you for the kind words! I'd really love to make the profile building part more accessible; less work for me too ;)

About the new bulbs, I would try it just in case, and maybe other similar WW/CW bulbs like the LSC 970716, 970715 and 970717 since sometimes they do match just fine.

ihrapsa, 2022-04-23:
Sure, will do some more tests once i get back home next week. :)

Also, I've seen you've renamed it to B27 but the bulb has an E27 screw mount, was that right!

Thanks again!

khalednassar, 2022-04-23:
> Sure, will do some more tests once i get back home next week. :)
> 
> Also, I've seen you've renamed it to B27 but the bulb has an E27 screw mount, was that right?
> 
> Thanks again!

Ah I thought it was B27 based on the issue title. No matter, quick revert will fix that ;)

ihrapsa, 2022-04-29:
> Thank you for the kind words @ihrapsa ! I'd really love to make the profile building part more accessible; less work for me too ;) It's definitely in the works, just taking a bit of time.
> 
> About the new bulbs, I would try it just in case, and maybe other similar WW/CW bulbs like the LSC 970716, 970715 and 970717 since sometimes they do match just fine.

Hi,
Unfortunately I haven't managed to find a working profile for the bulbs that were not upgraded to v1.1.4. These bulbs come on v1.0.8. So I dumped another bin. 🙂 

[ledvance_ww-cw_v1.0.8_dump.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8595130/ledvance_ww-cw_v1.0.8_dump.bin.zip)

MoogleTroupe, 2022-04-29:
is the profile working for the vendor fw version 1.1.4 bulbs?

ihrapsa, 2022-04-29:
> 
the newly added LEDVANCE profile works on the 1.1.4 bulbs but not on newly unboxed ones (which come on 1.0.8)

MoogleTroupe, 2022-04-29:
yeah it's indeed different. 95be7 is the payload target on that 1.0.8 fw version, looking for the register values. may have a profile for you soon.

MoogleTroupe, 2022-04-29:
pull latest and give the 1.0.8 profile a try.

ihrapsa, 2022-04-29:
It works. You guys are awesome! 😄 
Thank you!!

closed 2022-04-29 by ihrapsa