tuya-cloudcutter/tuya-cloudcutter #73: WB3S - Treatlife DS03 Smart Ceiling Fan Control and Dimmer Light Switch Dump

DatCFC, 2022-04-23:
WB3S: https://developer.tuya.com/en/docs/iot/wb3s-module-datasheet?id=K9dx20n6hz5n4

Treatlife DS03 fan/light switch:
https://www.treatlife.tech/collections/smart-dimmer-switch/products/smart-ceiling-fan-control-and-light-dimmer-switch-neutral-wire-needed-treatlife-single-pole-wi-fi-light-switch-fan-speed-control-works-with-alexa-google-assistant-schedule-remote-control?variant=31727498625085

https://www.amazon.com/Ceiling-Control-Treatlife-Assistant-Schedule/dp/B086PPRWL7

Dumped with bk7231tools
[treatlife_ds03.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8546062/treatlife_ds03.zip)

khalednassar, 2022-04-23:
Profile added to main, please check if it works for you.

N.B. the app code matches MartinJerry and the Treatlife DS02F, DPID schemas are quite different though

DatCFC, 2022-04-23:
Looks like it worked as I obtained the generated keys in the configured-devices folder

closed 2022-04-23 by MoogleTroupe