tuya-cloudcutter/tuya-cloudcutter #75: eMylo XL9251WI Smart Switch - Need help dumping

leecher1337, 2022-04-27:
CPU [CB2S](https://developer.tuya.com/en/docs/iot/cb2s-module-datasheet?id=Kafgfsa2aaypq)
PCB: 
[image]

I tried to connect 
VCC -> VCC, 
GND -> GND, 
TX -> RX1
RX -> TX1

and dump with:
```
pipenv run python uartprogram --unprotect -d com6 -r --startaddr 0x0 dump.bin
```

And also with 

```
pipenv run python bk7231tools.py read_flash -d COM6 -s 0 -c 512 dump.bin
```

But to no avail, device just doesn't react, even tough it gets power and LE Dlights up. Any thougts? I guess I won't be able to dump it without disconnecting RX1 from PCB - thus, destroying the device?
I can't access CEN because I cannot hold more connects in place simultanously, but I guess it also won't work, as it's basically the same as resetting VCC/GND for a short period of time?

MoogleTroupe, 2022-04-27:
It will be a little difficult if you cannot access CEN, and this also will depend on if your usb ttl can handle this procedure, however give this a try:

1. wire everything up, with GND having a momentary toggle switch on it.
2. Start the kb7231tools based dump. You should see your TX light flashing fast on the ttl device
3. immediately bound the ground pin such that the device reboots without disconnecting your TTL
4. You should see the RX light flash once, pause and start flashing if bk7231tools caught the bootloader in time. it is a *very* small window of time to catch the bootloader (the spamming TX is sending the "i want you in bootloader mode").
5. upload the resulting files here if successful. 

unlike chips like the ESP, which stay in bootloader mode, the only way on these beken chips is by it getting an inbound message at the right time on initial power up to enter bootloader mode.

MoogleTroupe, 2022-04-27:
I highly recommend a quality TTL on linux using bk7231tools. I have dumped several devices by power cycling gnd and not even using cen for the process like this.

tony-fav, 2022-04-27:
I've done all my dumping from Windows. A permission error means you have the wrong COM port or something else already is capturing that COM port before the script does.

openshwprojects, 2022-04-27:
there are many things you can do better and there is really no need to "destroy" the device

1. first of all, what do you mean by "vcc to vcc"? Beken is 3.3V and USB is 5V. You should connect 5V to AMS1117-3.3V input pin or use external LDO like TC1264
2. even if you power board correctly, you want to power it on and off to reboot device and this will take too much current and reset USB port. The solution is to connect GND and RX and TX from your USB port and get 5V elsewhere (other port or from USB charger). Ofc ground is common
3. are you using 3.3V logic levels on usb to uart converter?
4. if there is something connected on RX pin, you can easy cut the trace on the board , make sure with multimeter it's cut, and do a solder bridge (or bridge with cut of leg of resistor) to fix it after flashing
5. if you dont want to cut, then you can easily desolder whole board with the little help of solder wick and flux..

also "I can't access CEN because I cannot hold more connects in place simultanously,"? Are you trying to do that without soldering at all? If so, then I can clearly see how it can not be easy, but for soldering  you'll do ok with the cheapest soldering iron from china

MoogleTroupe, 2022-04-27:
if you use an external 5v power supply make sure you only have rx, tx, and gnd hooked to your ttl device. and reset the 5v line (leaving gnd connected) to try to get it to catch the bootloader message.

leecher1337, 2022-04-27:
ok, using a stable 3.3V power source helped with the "access denied" issues (device falling off USB bus), but the device still always boots up instead of entering boot loader mode, so my guess is that it's related to the RX line and my only optoin would be to find someone capable of soldering who  can cut it off and reattach it.
Well, at least I tried.

leecher1337, 2022-04-27:
Good question, I posted pretty much all relevant parts of the PCB above. The other side of the PCB doesn't have traces. the 3 solder points above R2 are the LED, can you spot something?

leecher1337, 2022-04-28:
My *guess* is that RX1 is connected to R3 and then to the LED and TX1 is connected to GND via the 000 solder bridge.
So I would need to cut both connections then? Or just RX1 to get rid of the resistor? But as said, that's just a guess.

MoogleTroupe, 2022-04-28:
you probably need to cut both, then reattach after the dump is completed. tx->gnd is for sure breaking things.

openshwprojects, 2022-04-28:
It would be really really really easier to desolder the board with module.
Flux + solder wick + soldering iron with clean tip.
I might try making a tutorial (video tutorial or step by step guide with photos?) of that...

leecher1337, 2022-04-28:
If you can make an idiot-proof tutorial for it, I'd be happy to prove that there will be a better idiot ;-)

closed 2022-12-09 by Cossid

leecher1337, 2023-02-22:
Updated dump file, now should be correct.
[emylo2-dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/10804426/emylo2-dump.zip)

Cossid, 2023-02-22:
Thanks!  A new device has been added as `eMylo XL9251WI Smart Switch`