tuya-cloudcutter/tuya-cloudcutter #77: DS-102-2 Wi-Fi 2gang Light Switch

leecher1337, 2022-04-28:
Product: [DS-102-2 Wi-Fi 2gang Light Switch](https://de.aliexpress.com/item/1005001698977140.html)
CPU: [WB3S](https://developer.tuya.com/en/docs/iot/wb3s-module-datasheet?id=K9dx20n6hz5n4)
Dump:
[dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8583418/dump.zip)

khalednassar, 2022-04-29:
Profile added on main branch. It actually matches another one, the Lonsonho device, so I presume this is just generic stuff. Could you give it a shot and see if it works for you?

leecher1337, 2022-04-29:
Hm, I cannot get to the point where it should connect to the cloudcutterflash AP. It may be my setup, because there are some strange error messages, but shouldn't I at least receive some connection attempt?
```
pi@iobroker-pi:~/tuya-cloudcutter $ sudo ./run_flash.sh wlan0 MINCO_HEAT/DS-102-2-Gang-WiFi-Switch
Building cloudcutter docker image..
Sending build context to Docker daemon  36.25MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 6b557201f18f
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 579561b057dd
Step 3/10 : FROM base AS python-deps
 ---> 579561b057dd
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> e9cc68880970
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 4c68b7a59d0b
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 3b467e9fac62
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> a71c52131eea
Step 8/10 : FROM python-deps AS cloudcutter
 ---> a71c52131eea
Step 9/10 : COPY src /src
 ---> Using cache
 ---> d317bc225d0d
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> c8044a181c4e
Successfully built c8044a181c4e
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231T_UG_1.0.16.bin
 > OpenBK7231T_UG_1.0.16.bin

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds.
Failed to start network-manager.service: Unit network-manager.service not found.
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Found access point name: "A-9D5C", trying to connect..
Device 'wlan0' successfully activated with '034a1969-1f4b-4d99-8192-c57963f78e5d'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/MAGvkxQqOGu2.deviceconfig
Saved device config in /work/configured-devices/MAGvkxQqOGu2.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds.
Failed to start network-manager.service: Unit network-manager.service not found.
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Found access point name: "A-9D5C", trying to connect..
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
Failed to start network-manager.service: Unit network-manager.service not found.
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
[... 14 lines cut ...]
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
^CTraceback (most recent call last):
  File "/usr/local/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/src/cloudcutter/__main__.py", line 340, in <module>
    args.handler(args)
  File "/src/cloudcutter/__main__.py", line 192, in __update_firmware
    __configure_local_device_or_update_firmware(args, update_firmare=True)
  File "/src/cloudcutter/__main__.py", line 160, in __configure_local_device_or_update_firmware
    tornado.ioloop.IOLoop.current().start()
[... cut ...] , in start
    self.asyncio_loop.run_forever()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 601, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1869, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/local/lib/python3.9/selectors.py", line 469, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
Oh no, something went wrong with updating firmware! Try again I guess..

```

MoogleTroupe, 2022-04-29:
you have a dns server running on your linux machine. please disable that, as there is one that has to be run inside the docker container in step 3 that is emulating some tuya cloud api calls.

leecher1337, 2022-04-29:
That solved DHCP issue, but still no connetion to the AP.. Hm

```
Successfully built c8044a181c4e
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select your custom firmware file: OpenBK7231T_UG_1.0.16.bin
 > OpenBK7231T_UG_1.0.16.bin

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds.
Failed to start network-manager.service: Unit network-manager.service not found.
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Found access point name: "A-9D5C", trying to connect..
Device 'wlan0' successfully activated with '034a1969-1f4b-4d99-8192-c57963f78e5d'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/CzISE3pV83GH.deviceconfig
Saved device config in /work/configured-devices/CzISE3pV83GH.deviceconfig
==> Turn the device off and on again once. Repeat 6 more times if your device's SSID doesn't show up within ~5 seconds.
Failed to start network-manager.service: Unit network-manager.service not found.
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Found access point name: "A-9D5C", trying to connect..
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
Failed to start network-manager.service: Unit network-manager.service not found.
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Scanning for "A-"  "SmartLife-"  "iHome-"  "Merkury-"  "Geeni-"  "GRID-"  "TreatLife-SL-"  "SL-CreeLighting-"  "LDV SMART+-" SSID...
Found access point name: "A-9D5C", trying to connect..
Device 'wlan0' successfully activated with '034a1969-1f4b-4d99-8192-c57963f78e5d'.
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: wlan0
Apr 29 15:14:51 dnsmasq[15]: started, version 2.80 cachesize 150
Apr 29 15:14:51 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Apr 29 15:14:51 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Apr 29 15:14:51 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlan0
Apr 29 15:14:51 dnsmasq[15]: read /etc/hosts - 5 addresses
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr 00:0f:00:3a:14:07 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
```

And then it waits forever at this point

leecher1337, 2022-04-29:
How long does it normally take after DHCP request and offt to continue?

```
Using interface wlan0 with hwaddr 00:0f:00:3a:14:07 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
Apr 29 15:15:50 dnsmasq-dhcp[15]: 440222768 available DHCP range: 10.42.42.10 -- 10.42.42.40
Apr 29 15:15:50 dnsmasq-dhcp[15]: 440222768 client provides name: wlan0
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 DHCPDISCOVER(wlan0) 38:1f:8d:57:9d:5c
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 tags: wlan0
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 DHCPOFFER(wlan0) 10.42.42.28 38:1f:8d:57:9d:5c
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 next server: 10.42.42.1
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  1 option: 53 message-type  2
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 54 server-identifier  10.42.42.1
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 51 lease-time  12h
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 58 T1  6h
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 59 T2  10h30m
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option:  1 netmask  255.255.255.0
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 28 broadcast  10.42.42.255
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option:  3 router  10.42.42.1
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option:  6 dns-server  10.42.42.1
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 available DHCP range: 10.42.42.10 -- 10.42.42.40
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 client provides name: wlan0
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 DHCPDISCOVER(wlan0) 38:1f:8d:57:9d:5c
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 tags: wlan0
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 DHCPOFFER(wlan0) 10.42.42.28 38:1f:8d:57:9d:5c
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 next server: 10.42.42.1
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  1 option: 53 message-type  2
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 54 server-identifier  10.42.42.1
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 51 lease-time  12h
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 58 T1  6h
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 59 T2  10h30m
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option:  1 netmask  255.255.255.0
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 28 broadcast  10.42.42.255
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option:  3 router  10.42.42.1
Apr 29 15:15:53 dnsmasq-dhcp[15]: 440222768 sent size:  4 option:  6 dns-server  10.42.42.1
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 available DHCP range: 10.42.42.10 -- 10.42.42.40
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 client provides name: wlan0
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 DHCPREQUEST(wlan0) 10.42.42.28 38:1f:8d:57:9d:5c
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 tags: wlan0
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 DHCPACK(wlan0) 10.42.42.28 38:1f:8d:57:9d:5c wlan0
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 next server: 10.42.42.1
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 sent size:  1 option: 53 message-type  5
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 54 server-identifier  10.42.42.1
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 51 lease-time  12h
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 58 T1  6h
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 59 T2  10h30m
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 sent size:  4 option:  1 netmask  255.255.255.0
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 sent size:  4 option: 28 broadcast  10.42.42.255
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 sent size:  4 option:  3 router  10.42.42.1
Apr 29 15:15:54 dnsmasq-dhcp[15]: 440222768 sent size:  4 option:  6 dns-server  10.42.42.1
```

leecher1337, 2022-04-29:
Hm 5 Minutes passed, now there is another request that gets ACKed. Shouldn't the device be happy with the ACKed lease request?

MoogleTroupe, 2022-04-29:
it should, but some devices can be finicky. I would try recutting a second time.

leecher1337, 2022-04-29:
With this device, entering AP mode works by pressing button 1 for 10 seconds, then it stays in AP mode. So no need for power cycling

leecher1337, 2022-04-29:
Thanks for the hint, will try

leecher1337, 2022-04-29:
Looks better now, but I guess it should download the firmware, I don't think it progresses:

```
Apr 29 15:33:34 dnsmasq-dhcp[16]: 1763700041 sent size:  4 option:  3 router  10.42.42.1
Apr 29 15:33:34 dnsmasq-dhcp[16]: 1763700041 sent size:  4 option:  6 dns-server  10.42.42.1
Apr 29 15:34:10 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:34:10 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:34:11 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:34:11 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:34:12 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:34:12 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:34:14 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:34:14 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:34:42 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:34:42 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:34:43 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:34:43 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:34:44 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:34:44 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:34:46 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:34:46 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:35:14 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:35:14 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:35:15 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:35:15 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:35:16 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:35:16 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:35:18 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:35:18 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:35:29 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:35:29 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:35:30 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:35:30 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:35:31 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:35:31 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:35:33 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:35:33 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:36:02 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:36:02 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:36:02 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:36:02 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:36:03 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:36:03 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
Apr 29 15:36:05 dnsmasq[16]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 15:36:05 dnsmasq[16]: config h3.iot-dns.com is 10.42.42.1
```

```
root@iobroker-pi:/var/log# netstat -antp | grep 80
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      26255/python3
tcp        0      0 10.42.42.1:80           10.42.42.1:55180        TIME_WAIT   -
tcp6       0      0 :::80                   :::*                    LISTEN      26255/python3
root@iobroker-pi:/var/log# ps -ef | grep python
root     26207 25049  0 16:33 pts/1    00:00:00 docker run --network=host -ti --privileged -v /home/pi/tuya-cloudcutter:/work cloudcutter bash -c bash /src/setup_apmode.sh wlan0 && pipenv run python3 -m cloudcutter update_firmware "/work/device-profiles/MINCO_HEAT/DS-102-2-Gang-WiFi-Switch" "/work/configured-devices/CNFUIojJUR3t.deviceconfig" "/work/custom-firmware/OpenBK7231T_UG_1.0.16.bin"
root     26255 26233 11 16:33 pts/0    00:00:23 /src/.venv/bin/python3 -m cloudcutter update_firmware /work/device-profiles/MINCO_HEAT/DS-102-2-Gang-WiFi-Switch /work/configured-devices/CNFUIojJUR3t.deviceconfig /work/custom-firmware/OpenBK7231T_UG_1.0.16.bin
root     26638 26539  0 16:36 pts/3    00:00:00 grep python
```

So I have a TIME_WAIT connection from the client, at least, but that's it. Is there a log that show access to HTTP port 80?

MoogleTroupe, 2022-04-29:
do you have the latest copy of this repo cloned?
have you tested if a normal cut from cloud but use tuya api works?

that error message i've seen in the past for some devices that were using "different" api endpoints. but if i recall, that edge case is now covered, though it is certainly possibly this is yet another edge case.

leecher1337, 2022-04-29:
Yes, using latest repo (otherwise, I wouldn't have device profile). And have not tested "normal" cloud cut. I can try it out, too.. So it would still be possible to flash firmware with the same script after cloud cut?

khalednassar, 2022-04-29:
This might be indeed yet another edge case. I've just pushed a change for more verbose HTTP output, which would help debug the issue.

leecher1337, 2022-04-29:
Hm, that's interesting:
```
17:02:10.655422 IP 0.0.0.0.bootpc > 255.255.255.255.bootps: BOOTP/DHCP, Request from 38:1f:8d:57:9d:5c (oui Unknown), length 308
17:02:12.580827 IP 0.0.0.0.bootpc > 255.255.255.255.bootps: BOOTP/DHCP, Request from 38:1f:8d:57:9d:5c (oui Unknown), length 308
17:02:13.686497 IP 10.42.42.1.bootps > 10.42.42.28.bootpc: BOOTP/DHCP, Reply, length 300
17:02:13.693897 IP 10.42.42.1.bootps > 10.42.42.28.bootpc: BOOTP/DHCP, Reply, length 300
17:02:13.700793 IP 0.0.0.0.bootpc > 255.255.255.255.bootps: BOOTP/DHCP, Request from 38:1f:8d:57:9d:5c (oui Unknown), length 308
17:02:13.716424 IP 10.42.42.1.bootps > 10.42.42.28.bootpc: BOOTP/DHCP, Reply, length 300
17:02:13.722671 ARP, Request who-has 10.42.42.28 tell 10.42.42.28, length 28
17:02:14.558038 ARP, Request who-has 10.42.42.1 tell 10.42.42.28, length 28
17:02:14.558149 ARP, Reply 10.42.42.1 is-at 00:0f:00:3a:14:07 (oui Unknown), length 28
17:02:14.603500 IP 10.42.42.28.53527 > 52.224.135.238.https: Flags [S], seq 6509, win 4380, options [mss 1460], length 0
17:02:16.307210 IP 10.42.42.28.53527 > 52.224.135.238.https: Flags [S], seq 6509, win 4380, options [mss 1460], length 0
17:02:18.306118 IP 10.42.42.28.53527 > 52.224.135.238.https: Flags [S], seq 6509, win 4380, options [mss 1460], length 0
17:02:18.721422 ARP, Request who-has 10.42.42.28 tell 10.42.42.1, length 28
17:02:18.800686 ARP, Reply 10.42.42.28 is-at 38:1f:8d:57:9d:5c (oui Unknown), length 28
17:02:20.305955 IP 10.42.42.28.53527 > 52.224.135.238.https: Flags [S], seq 6509, win 4380, options [mss 1460], length 0
17:02:22.305740 IP 10.42.42.28.53527 > 52.224.135.238.https: Flags [S], seq 6509, win 4380, options [mss 1460], length 0
17:02:24.437864 IP 10.42.42.28.53527 > 52.224.135.238.https: Flags [S], seq 6509, win 4380, options [mss 1460], length 0
17:02:26.305206 IP 10.42.42.28.53527 > 52.224.135.238.https: Flags [S], seq 6509, win 4380, options [mss 1460], length 0
17:02:28.507483 IP 10.42.42.28.53527 > 52.224.135.238.https: Flags [S], seq 6509, win 4380, options [mss 1460], length 0
17:02:30.306359 IP 10.42.42.28.53527 > 52.224.135.238.https: Flags [S], seq 6509, win 4380, options [mss 1460], length 0
17:02:31.987009 IP 10.42.42.28.59618 > 10.42.42.1.domain: 42877+ A? h3.iot-dns.com. (32)
17:02:32.158433 IP 10.42.42.28.59618 > 10.42.42.1.domain: 42877+ A? h3.iot-dns.com. (32)
17:02:33.158900 IP 10.42.42.28.59618 > 10.42.42.1.domain: 42877+ A? h3.iot-dns.com. (32)
17:02:35.162020 IP 10.42.42.28.59618 > 10.42.42.1.domain: 42877+ A? h3.iot-dns.com. (32)
17:02:38.478480 IP 10.42.42.28.59618 > 192.168.175.1.domain: 42877+ A? h3.iot-dns.com. (32)
17:02:39.281092 IP 10.42.42.28.59618 > 192.168.175.1.domain: 42877+ A? h3.iot-dns.com. (32)
17:02:40.162434 IP 10.42.42.28.59618 > 192.168.175.1.domain: 42877+ A? h3.iot-dns.com. (32)
17:02:42.076782 IP 10.42.42.28.59618 > 192.168.175.1.domain: 42877+ A? h3.iot-dns.com. (32)
17:02:46.315350 IP 10.42.42.28.53528 > ec2-34-218-107-222.us-west-2.compute.amazonaws.com.https: Flags [S], seq 6542, win 4380, options [mss 1460], length 0
17:02:48.268705 IP 10.42.42.28.53528 > ec2-34-218-107-222.us-west-2.compute.amazonaws.com.https: Flags [S], seq 6542, win 4380, options [mss 1460], length 0
17:02:50.268606 IP 10.42.42.28.53528 > ec2-34-218-107-222.us-west-2.compute.amazonaws.com.https: Flags [S], seq 6542, win 4380, options [mss 1460], length 0
17:02:52.508768 IP 10.42.42.28.53528 > ec2-34-218-107-222.us-west-2.compute.amazonaws.com.https: Flags [S], seq 6542, win 4380, options [mss 1460], length 0
17:02:54.268981 IP 10.42.42.28.53528 > ec2-34-218-107-222.us-west-2.compute.amazonaws.com.https: Flags [S], seq 6542, win 4380, options [mss 1460], length 0
17:02:56.487178 IP 10.42.42.28.53528 > ec2-34-218-107-222.us-west-2.compute.amazonaws.com.https: Flags [S], seq 6542, win 4380, options [mss 1460], length 0
17:02:58.228422 IP 10.42.42.28.53528 > ec2-34-218-107-222.us-west-2.compute.amazonaws.com.https: Flags [S], seq 6542, win 4380, options [mss 1460], length 0
17:03:00.463521 IP 10.42.42.28.53528 > ec2-34-218-107-222.us-west-2.compute.amazonaws.com.https: Flags [S], seq 6542, win 4380, options [mss 1460], length 0
17:03:02.228792 IP 10.42.42.28.53528 > ec2-34-218-107-222.us-west-2.compute.amazonaws.com.https: Flags [S], seq 6542, win 4380, options [mss 1460], length 0
17:03:03.993649 IP 10.42.42.28.23655 > 10.42.42.1.domain: 48469+ A? h3.iot-dns.com. (32)
17:03:04.255894 IP 10.42.42.28.23655 > 10.42.42.1.domain: 48469+ A? h3.iot-dns.com. (32)
17:03:05.213115 IP 10.42.42.28.23655 > 10.42.42.1.domain: 48469+ A? h3.iot-dns.com. (32)
17:03:07.248144 IP 10.42.42.28.23655 > 10.42.42.1.domain: 48469+ A? h3.iot-dns.com. (32)
```

khalednassar, 2022-04-29:
@leecher1337 this is pretty odd output. It shouldn't even be able to resolve these domain names let alone try to actually connect. Are you absolutely sure nothing else is responding to its DNS queries? The provided `dnsmasq` configuration should respond with 10.42.42.1 for all DNS queries.

leecher1337, 2022-04-29:
It says that it does, but what puzzles me is that I don'T see this reply in tcpdump...?
```
Apr 29 16:03:36 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Apr 29 16:03:37 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 16:03:37 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Apr 29 16:03:39 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 16:03:39 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Apr 29 16:04:08 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 16:04:08 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Apr 29 16:04:08 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 16:04:08 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Apr 29 16:04:09 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 16:04:09 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Apr 29 16:04:11 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.28
Apr 29 16:04:11 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
```

leecher1337, 2022-04-29:
I'll clean up the iptables, maybe it's a firewall issue on this box, otherwise I'll buy a second SD card and try with a completely clean RaspPi before botherign you with it again.

khalednassar, 2022-04-29:
Nothing jumps out to me immediately. The docker chain shouldn't really apply anyway and the container is running in the host's network ns so not sure why it'd go off.

Maybe also check the kernel message buf?

leecher1337, 2022-04-29:
Nothing strange in the logs so far. But it must be an issue with the machine if the packet is not leaving the docker container. The DHCP (BOOTP) traffic is leaving the container without issues. hm
Well, I'll get a new SD card then, maybe the RasPi image is broken somehow.

leecher1337, 2022-05-01:
Ok, tried again with a clean Raspian image.
2 things I had to fix:

1) Package is called NetworkManager on Raspi, so I had to replace "network-manager" with "NetworkManager" in common.sh and run_detach.sh
2) common.sh: nmcli -f GENERAL.STATE looks for "activated", but that's a localized string, so on my german system, it said "aktiviert". I fixed it by prepending LANG=en_US.UTF-8 to nmcli:
```
AP_STATUS=$(LANG=en_US.UTF-8 nmcli -f GENERAL.STATE con show "${AP_MATCHED_NAME}" | awk -F ' ' '{print $2}')
```

It then ran through in my attempt to Install OpenBK firmware. I don't have an OpenBK WiFi now, so something may not work, but maybe OpenBK just isn't supported on the device...?
Here is the log:
```
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:31:fe:e7 and ssid "cloudcutterflash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
May  1 09:09:20 dnsmasq-dhcp[15]: 899820720 available DHCP range: 10.42.42.10 -- 10.42.42.40
May  1 09:09:20 dnsmasq-dhcp[15]: 899820720 client provides name: wlan0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 DHCPDISCOVER(wlan0) 38:1f:8d:57:9d:5c
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 tags: wlan0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 DHCPOFFER(wlan0) 10.42.42.28 38:1f:8d:57:9d:5c
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 next server: 10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  1 option: 53 message-type  2
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 54 server-identifier  10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 51 lease-time  12h
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 58 T1  6h
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 59 T2  10h30m
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option:  1 netmask  255.255.255.0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 28 broadcast  10.42.42.255
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option:  3 router  10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option:  6 dns-server  10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 available DHCP range: 10.42.42.10 -- 10.42.42.40
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 client provides name: wlan0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 DHCPDISCOVER(wlan0) 38:1f:8d:57:9d:5c
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 tags: wlan0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 DHCPOFFER(wlan0) 10.42.42.28 38:1f:8d:57:9d:5c
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 next server: 10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  1 option: 53 message-type  2
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 54 server-identifier  10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 51 lease-time  12h
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 58 T1  6h
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 59 T2  10h30m
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option:  1 netmask  255.255.255.0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 28 broadcast  10.42.42.255
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option:  3 router  10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option:  6 dns-server  10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 available DHCP range: 10.42.42.10 -- 10.42.42.40
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 client provides name: wlan0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 DHCPREQUEST(wlan0) 10.42.42.28 38:1f:8d:57:9d:5c
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 tags: wlan0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 DHCPACK(wlan0) 10.42.42.28 38:1f:8d:57:9d:5c wlan0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 next server: 10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  1 option: 53 message-type  5
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 54 server-identifier  10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 51 lease-time  12h
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 58 T1  6h
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 59 T2  10h30m
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option:  1 netmask  255.255.255.0
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option: 28 broadcast  10.42.42.255
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option:  3 router  10.42.42.1
May  1 09:09:23 dnsmasq-dhcp[15]: 899820720 sent size:  4 option:  6 dns-server  10.42.42.1
May  1 09:09:58 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.28
May  1 09:09:58 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 01
[I 220501 09:09:59 web:2239] 200 POST /v1/url_config (10.42.42.28) 81.02ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.active&et=1&t=61&uuid=hCDjuk2KU62w&v=4.4&sign=0403c3dce49eb24ee2078c97886f78b8
[... 89 lines cut ...]
[... cut ...]
[LOG] Client request PATH: /d.json - QS: a=tuya.device.uuid.pskkey.get&et=1&t=1651396235&uuid=hCDjuk2KU62w&v=1.0&sign=0a24d580ee7ca16ed89e3948e699bf77
[LOG] ==== Client request body ===
{'t': 1651396235}
[LOG] ==== End client request body ===
tuya.device.uuid.pskkey.get - response: {'result': {'pskKey': ''}, 't': 1651396236, 'success': True}
[I 220501 09:10:36 web:2239] 200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=1651396235&uuid=hCDjuk2KU62w&v=1.0&sign=0a24d580ee7ca16ed89e3948e699bf77 (10.42.42.28) 99.45ms
```

leecher1337, 2022-05-01:
When trying to flash OpenBK firmware, it quits at a certain stage, not sure, if this is intended?
```
Using PSK v1 - Received PSK ID version 01
[I 220501 09:41:58 web:2239] 200 POST /v1/url_config (10.42.42.28) 24.09ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.active&et=1&t=34&uuid=7srgVsgakRcx&v=4.4&sign=dc0eba1b4f083570f5c1d59269008fc5
[LOG] ==== Client request body ===
{'token': 'AAAAAA', 'softVer': '1.1.5', 'productKey': 'key7axydcvmea3x9', 'protocolVer': '2.2', 'baselineVer': '40.00', 'productKeyStr': 'key7axydcvmea3x9', 'devId': 'XgFvWpTpThmp1btHp1GF', 'hid': '381f8d579d5c', 'modules': '[{"type":9,"softVer":"1.1.5","online":true}]', 'devAttribute': 515, 'cadVer': '1.0.2', 'cdVer': '1.0.0', 'options': '{"isFK":true}', 't': 34}
[LOG] ==== End client request body ===
tuya.device.active - response: {'result': {'schema': '[{"type":"obj","mode":"rw","property":{"type":"bool"},"id":1},{"type":"obj","mode":"rw","property":{"type":"bool"},"id":2},{"type":"obj","mode":"rw","property":{"min":0,"max":86400,"scale":0,"step":1,"type":"value"},"id":7},{"type":"obj","mode":"rw","property":{"min":0,"max":86400,"scale":0,"step":1,"type":"value"},"id":8},{"type":"obj","mode":"rw","property":{"range":["off","on","memory"],"type":"enum"},"id":14},{"type":"obj","mode":"rw","property":{"type":"string","maxlen":255},"id":17},{"type":"obj","mode":"rw","property":{"type":"string","maxlen":255},"id":18},{"type":"obj","mode":"rw","property":{"type":"string","maxlen":255},"id":19}]', 'devId': 'BFCFsH21s3vriNY0sJqB', 'resetFactory': False, 'timeZone': '+02:00', 'capability': 1025, 'secKey': 'FcFDaIpyxalnnZD5', 'stdTimeZone': '+01:00', 'schemaId': '0000030tuw', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200], [1711846800, 1729990800], [1743296400, 1761440400], [1774746000, 1792890000]], 'localKey': '34IYDHutrmeK0MfV'}, 't': 1651398119, 'success': True}
[I 220501 09:41:59 web:2239] 200 POST /d.json?a=tuya.device.active&et=1&t=34&uuid=7srgVsgakRcx&v=4.4&sign=dc0eba1b4f083570f5c1d59269008fc5 (10.42.42.28) 123.72ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.dynamic.config.get&devId=BFCFsH21s3vriNY0sJqB&et=1&t=1651398119&v=1.0&sign=6c02750f76d9d2f9297e38620e4ae889
[LOG] ==== Client request body ===
{'t': 1651398119}
[LOG] ==== End client request body ===
tuya.device.dynamic.config.get - response: {'result': {'ackId': '0-0', 'validTime': 1800, 'time': 1651398120, 'config': {'stdTimeZone': '+01:00', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200]]}}, 't': 1651398120, 'success': True}
[I 220501 09:42:00 web:2239] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=BFCFsH21s3vriNY0sJqB&et=1&t=1651398119&v=1.0&sign=6c02750f76d9d2f9297e38620e4ae889 (10.42.42.28) 148.80ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.uuid.pskkey.get&et=1&t=1651398122&uuid=7srgVsgakRcx&v=1.0&sign=3e811adc459dfa9c7bcef4bee439fe3c
[LOG] ==== Client request body ===
{'t': 1651398122}
[LOG] ==== End client request body ===
tuya.device.uuid.pskkey.get - response: {'result': {'pskKey': ''}, 't': 1651398123, 'success': True}
[I 220501 09:42:03 web:2239] 200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=1651398122&uuid=7srgVsgakRcx&v=1.0&sign=3e811adc459dfa9c7bcef4bee439fe3c (10.42.42.28) 96.52ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1651398125} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
[LOG] Client request PATH: /d.json - QS: a=tuya.device.upgrade.get&devId=BFCFsH21s3vriNY0sJqB&et=1&t=1651398125&v=4.3&sign=1709515cb2fcc739bc7c5b7d989cc596
[LOG] ==== Client request body ===
{'type': 0, 't': 1651398125}
[LOG] ==== End client request body ===
tuya.device.upgrade.get - response: {'result': {'url': 'http://10.42.42.1:80/files/OpenBK7231T_UG_1.0.19.bin', 'hmac': '518F08D860B1979922879A12C4B30CBE2FCD6736A0C55016AE720E16E36EC96F', 'version': '9.0.0', 'size': '368352', 'type': 0}, 'success': True, 't': 1651398126}
[I 220501 09:42:06 web:2239] 200 POST /d.json?a=tuya.device.upgrade.get&devId=BFCFsH21s3vriNY0sJqB&et=1&t=1651398125&v=4.3&sign=1709515cb2fcc739bc7c5b7d989cc596 (10.42.42.28) 14.62ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.upgrade.status.update&devId=BFCFsH21s3vriNY0sJqB&et=1&t=1651398126&v=4.1&sign=f8fa2bb4ac4ed0feeef41b258a49a861
[LOG] ==== Client request body ===
{'type': 0, 'upgradeStatus': 2, 't': 1651398126}
[LOG] ==== End client request body ===
tuya.device.upgrade.status.update - response: {'success': True, 't': 1651398126}
[I 220501 09:42:06 web:2239] 200 POST /d.json?a=tuya.device.upgrade.status.update&devId=BFCFsH21s3vriNY0sJqB&et=1&t=1651398126&v=4.1&sign=f8fa2bb4ac4ed0feeef41b258a49a861 (10.42.42.28) 23.99ms
[LOG] Client request PATH: /d.json - QS: a=atop.online.debug.log&devId=BFCFsH21s3vriNY0sJqB&et=1&t=1651398129&sign=4c874c0a133ff3e0d6d671a403f8649f
[LOG] ==== Client request body ===
{'data': '4', 't': 1651398129}
[LOG] ==== End client request body ===
atop.online.debug.log - response: {'result': True, 't': 1651398129, 'success': True}
[I 220501 09:42:09 web:2239] 200 POST /d.json?a=atop.online.debug.log&devId=BFCFsH21s3vriNY0sJqB&et=1&t=1651398129&sign=4c874c0a133ff3e0d6d671a403f8649f (10.42.42.28) 17.33ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.dynamic.config.get&devId=BFCFsH21s3vriNY0sJqB&et=1&t=1651398129&v=1.0&sign=cee460bdf610cf8605f83ebf1d642b9b
[LOG] ==== Client request body ===
{'t': 1651398129}
[... cut ...]
[I 220501 09:42:21 web:2239] 206 GET /files/OpenBK7231T_UG_1.0.19.bin (10.42.42.28) 156.60ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=101847-368351/368352
[I 220501 09:42:26 web:2239] 206 GET /files/OpenBK7231T_UG_1.0.19.bin (10.42.42.28) 326.92ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=101847-368351/368352
[I 220501 09:42:32 web:2239] 206 GET /files/OpenBK7231T_UG_1.0.19.bin (10.42.42.28) 221.57ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=101847-368351/368352
pi@raspberrypi:~/tuya-cloudcutter $
```

khalednassar, 2022-05-01:
It seems that it's downloading the update just fine, but perhaps the exit timeout is too aggressive or the config messages are off for your device. Could you try changing the timeout on this line https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/26fcb5746b23302725c18934e0ed896434cf97dc/src/cloudcutter/__main__.py#L81 from 30.0 to something higher, say 60?

leecher1337, 2022-05-01:
Interesting.. Set timeout to 60 sec and retried:

```
[I 220501 20:40:09 web:2239] 200 POST /d.json?a=atop.online.debug.log&devId=HDaBaJGSmv7TlrH0Ic4e&et=1&t=1651437609&sign=64e00d3a428d79c0a42d1d2a110e7609 (10.42.42.28) 8.74ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.dynamic.config.get&devId=HDaBaJGSmv7TlrH0Ic4e&et=1&t=1651437609&v=1.0&sign=1ba9d3c591bdae69cf05d574f10ee3a7
[LOG] ==== Client request body ===
{'t': 1651437609}
[LOG] ==== End client request body ===
tuya.device.dynamic.config.get - response: {'result': {'ackId': '0-0', 'validTime': 1800, 'time': 1651437609, 'config': {'stdTimeZone': '+01:00', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200]]}}, 't': 1651437609, 'success': True}
[I 220501 20:40:09 web:2239] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=HDaBaJGSmv7TlrH0Ic4e&et=1&t=1651437609&v=1.0&sign=1ba9d3c591bdae69cf05d574f10ee3a7 (10.42.42.28) 11.15ms
[LOG] Client request PATH: /d.json - QS: a=tuya.device.dynamic.config.ack&devId=HDaBaJGSmv7TlrH0Ic4e&et=1&t=1651437609&v=1.0&sign=e456d5b6bdf26e6b03d4023cce222b78
[LOG] ==== Client request body ===
{'ackId': '0-0', 't': 1651437609}
[LOG] ==== End client request body ===
tuya.device.dynamic.config.ack - response: {'t': 1651437610, 'success': True}
[I 220501 20:40:10 web:2239] 200 POST /d.json?a=tuya.device.dynamic.config.ack&devId=HDaBaJGSmv7TlrH0Ic4e&et=1&t=1651437609&v=1.0&sign=e456d5b6bdf26e6b03d4023cce222b78 (10.42.42.28) 12.71ms
[I 220501 20:40:12 web:2239] 200 GET /files/OpenBK7231T_UG_1.0.19.bin (10.42.42.28) 6343.52ms
[DEVICE OTA] Responding to device OTA HTTP request range: bytes=0-368351/368352
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1651437614} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
pi@raspberrypi:~/tuya-cloudcutter $
```
Now it didn't request firmware files and now when I plug in device, it just shows up the blue led and does nothing anymore.
Maybe it erased its flash and killed itself?

leecher1337, 2022-05-01:
Ok, forget it, firmware update worked, I was just confused because it lacked  OTA HTTP request range: bytes=101847-368351/368352 message.
So I can confirm that process works!

MoogleTroupe, 2022-05-01:
awesome great to hear!

closed 2022-05-01 by MoogleTroupe

leecher1337, 2022-05-01:
Here is a summary on how to flash OpenBK to the device for further reference:

1) Enter access point mode of device (indicated by LED of left button slowly blinking): Push and hold left button for 10 seconds until device resets and starts to blink slowly
2) Download repository on RaspPi and set it up according to [instructions](https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md)
3) Apply patches for [NetworkManager and language fix](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/77#issuecomment-1114183221)
4) Adjust [timeout](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/77#issuecomment-1114189490) 
5) Download current [OpenBK7231T_UG release](https://github.com/openshwprojects/OpenBK7231T_App):
```
cd custom-firmware/
 wget https://github.com/openshwprojects/OpenBK7231T_App/releases/download/1.0.19/OpenBK7231T_UG_1.0.19.bin
cd ..
```
5) Flash to device and follow instructions:
```
sudo ./run_flash.sh wlan0 MINCO_HEAT/DS-102-2-Gang-WiFi-Switch
```