tuya-cloudcutter/tuya-cloudcutter #80: Polish WB2L bulb "Inteligentna żarówka" LED LSC Smart Connect 10W 806L 970715.1 v1.1 from shop Action

openshwprojects, 2022-04-29:
A firmware from an Elektroda user, see this topic:
https://www.elektroda.pl/rtvforum/viewtopic.php?p=20000057#20000057
he has more of them and might check tuya-cloudcutter

Product page:
https://www.action.com/pl-pl/p/inteligentna-arowka-led-lsc-smart-connect/
Photo:
[image]
[bk7231s_dump-2022- 4-29-14-26-25.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8591195/bk7231s_dump-2022-.4-29-14-26-25.zip)

MoogleTroupe, 2022-04-29:
can you ask them to test both of the existing "3000272" lsc profiles, one of them may already match this bulb.

MoogleTroupe, 2022-04-29:
specifically, i see he mentioned it has 970715.1  on it, so the 3000272-970715 profile probably works.

openshwprojects, 2022-04-29:
@MoogleTroupe  I have forwarded your message and asked him to check it out, we will see what he says

MoogleTroupe, 2022-05-09:
change `[]` to `wlan0`

MoogleTroupe, 2022-05-09:
you also need to install network-manager, i suggest reading the raspbian readme https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md

dkliszewski, 2022-05-10:
before I start, I used this manual from this link and I installed everything (network-manager too)
systemctl:
`networking.service                                                                           loaded active exited    Raise network interfaces

NetworkManager-wait-online.service                                                           loaded active exited    Network Manager Wait Online

NetworkManager.service                                                                       loaded active running   Network Manager
`

closed 2022-11-21 by Cossid