tuya-cloudcutter/tuya-cloudcutter #81: CB3S BK7231N AVATTO Tuya WiFi Door Sensor, Smart Door Open/Closed Detectors model DS06

openshwprojects, 2022-04-29:
[image]

this device has tuyamcu but it can be converted to openbeken even by removing mcu

out of curiosity i tried reading flash in 3 methods, bkwriter 1.60 in two modes and hid_download_tool without crc check, but binaries seems similiar.
[image]

[image]

[image]

[CB3S_door_sensor.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8591920/CB3S_door_sensor.zip)

full review will be posted on Elektroda later

MoogleTroupe, 2022-04-29:
>but binaries seems similiar.

rolling hashes is because of the boot counter writing to flash changes the checksum of a full dump a little bit.

MkMunich, 2022-06-20:
Hi, my Tuya Wi-Fi door window sensor look very similar to this one: The LPC 2129 is 90 degrees
[image]

Could you please let me know, (for a newby) how to extract the binaries?
Best Regards

P.S. I would like to flash a firmware, 'free' from tuya server - best it would connect to (any) WiFi router of my own in order to provide its door state. Possible??

closed 2022-11-30 by Cossid