tuya-cloudcutter/tuya-cloudcutter #84: Add explanation in how-to

carefulcomputer, 2022-05-05:
In 
https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/HOST_SPECIFIC_INSTRUCTIONS.md 

step 5 under rasp pi 2w it says 

`5. Share your computers network with the Pi`

Can you please explain what that means ?

tony-fav, 2022-05-05:
There are instructions on doing that here https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget/ethernet-gadget

carefulcomputer, 2022-05-05:
I think your link points to instructions on how to connect to pi, those instructions precedes step 5. It will be nice to explain what does 'sharing the computer network with pi' means. Does it mean that pi needs to connect to wifi on local network  ?

tony-fav, 2022-05-05:
Apologies. Wrong link. https://desertbot.io/blog/headless-pi-zero-ssh-access-over-usb-windows "Step 10. Access the network"

If you plan to install packages and get the repo onto the pi, you need to do that some way, giving it a network connection is the easiest. You could also use the wifi to do all the setup.

carefulcomputer, 2022-05-08:
Another update. Setting up internet sharing with Mac is turning out to be way more difficult than I thought. I am able to connect to rpi using USB, but there are no clear step by step guide I could find to make internet sharing work with mac.
Looks like there are some files changed in recent version of rpi (related to dhcpcd.conf and network/interfaces).
My rpi Z w will not share connect, spent two days trying to reset rpi again and again. Would you happen to have a recent guide ?

Cossid, 2022-12-10:
Closing this ticket as stale.  Documentation has been significantly updated since this ticket was opened.

closed 2022-12-10 by Cossid