tuya-cloudcutter/tuya-cloudcutter #86: original firmware Qiachip 110V220V Light and Fan Controller (FLC-220VFLC-110V) with CB2S module

openshwprojects, 2022-05-08:
This is firmware from Jennifer, I am just posting it here.
Can anyone check if that's exploitable before we flash OpenBK on her device? 
Is this a correct dump?
[original firmware Qiachip 110V220V Light and Fan Controller (FLC-220VFLC-110V) with CB2S module.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8647522/original.firmware.Qiachip.110V220V.Light.and.Fan.Controller.FLC-220VFLC-110V.with.CB2S.module.zip)
[image]
[image]
[image]
[image]

MoogleTroupe, 2022-05-09:
The dump is good, exploit will be made later, feel free to let them know they can do what they need to now.

jenniferlee1818, 2022-05-11:
[image]

Cossid, 2022-12-01:
This has an unverified schema attached to a verified exploit.  Appears this device no longer has original firmware to get a proper dump, closing issue.  Re-open if a complete dump can be provided.

closed 2022-12-01 by Cossid