tuya-cloudcutter/tuya-cloudcutter #88: Treatlife SL10 - PSK ID 02

marcosrogers, 2022-05-10:
First off, this is a great project. Thank you so much for all the hard work!

When testing the exploit with my Treatlife SL10 bulbs, I ran into the following error, after which the exploit could not continue; it just tried the connection again:
`Using PSK v1 - Received PSK ID version 02` 
Here's a [full log file of the exploit attempt](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8665201/log-Treatlife-SL10-failed-obfuscated.txt). The error first occurs on line 138.

I believe my bulbs might be a slightly different model from the device profile that already exists (SL10-OMNI-TP120). The branding on this bulb says "TreatLife SL-10 **G1** TP120-9/A19," and it uses a WB8P chip (BK7231S_1.0.5).

Here's a [firmware dump](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8665116/SL10-dump.zip) I took after the failed attempt. Will it work for creating a new profile?

MoogleTroupe, 2022-05-11:
try the treatlife sl20 profile

marcosrogers, 2022-05-11:
I get the same info/error as before:
```
Using PSK v1 - Received PSK ID version 02
[W 220511 01:24:26 iostream:1406] SSL Error on 12 ('10.42.42.10', 57386): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
```

Here's the [log file](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8665941/log-Treatlife-SL20_Profile-failed-obfuscated.txt) using the SL 20 profile.

MoogleTroupe, 2022-05-11:
looks like a new app partition version, a new profile will be built for it soon

openshwprojects, 2022-05-12:
Would it be possible to somehow make profile generation more automatic?

mcblok, 2022-06-09:
has the new profile been built yet? I have the same bulbs

khalednassar, 2022-06-10:
Added the profile as TreatLife/SL10-v2 - can someone give it a shot?

mcblok, 2022-06-10:
Thanks for the quick response. I just tried and still get the SSL error:

SSL Error on 12 ('10.42.42.33', 51858): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)

marcosrogers, 2022-06-10:
Yes, using the Smart Life app. Most recently, I had them running with LocalTuya by signing up for a Tuya developer account and getting the local_key and device_id from there.

When running the exploit, I specified my own randomly-generated values for local key and device id.

Just to be sure, you're using the new SL10-v2 profile rather than the previous one, right?

mcblok, 2022-06-10:
Yes,I'm using the SL10-v2 profile. Also tried the other SL10 profile with the same results.I just registered the bulb with the Smart Life app and tried again but no bueno. Perhaps the firmware is different? Did you have to open the bulb to pull the firmware?

marcosrogers, 2022-06-10:
Yeah.. I had to open up the bulb to get the firmware dump

marcosrogers, 2022-08-10:
Sorry for the long delay in following up. I was out of town and unable to test these bulbs further.

Following up on my previous post...
> I've tested on two bulbs, and they both get to the line `Device should be successfully onboarded on WiFi AP!`
> 
> I can see them attach to wifi from my router logs, but so far I haven't gotten them to connect to Home Assistant using the LocalTuya integration.

The exploit seems to have worked with regard to setting new device keys and local keys. The bulbs also connect successfully to the WiFi. However, the datapoints seem to be inaccurate. Here is what I get when I try to set up the bulbs on localtuya
[image]

I'm guessing that the datapoints are defined when the exploit is run since Localtuya was able to connect to these bulbs using the keys I got from the Tuya API, before running the exploit. If they are defined by the exploit, it seems that the datapoints defined in the SL10-v2 profile should be different from the SL10-OMNI-TP120 profile, since the only differences I see in the profiles are the payload and datagram_padding. If they're not, I'm at a loss as to what to try next. Please advise.

kwiksilver, 2022-08-11:
So this profile in this link doesn't work for your bulb?
The part right here? 

Entity with DPS 20 and platform light.
Brightness 22 (value 1000)
Color Temp 23 (value 44)
Color Mode 21 (value white)
Color 24 (000003e803e8)
Scene 25 (000e0d0000000000000c0000)

marcosrogers, 2022-08-11:
The error appears before I'm even able to enter any of those values in Localtuya. The values I used successfully before I ran the exploit were:
- DPS 20
- Platform light
- Brightness: 22 (value: 1000)
- Color temperature: 23 (value: **220**)
- Brightness lower value: 29
- Brightness upper value: 1000
- Color mode: 21 (value: white)
- Color: 24 (color: **0115035c0320**)
- Minimum color temperature in K: 2700
- Maximum color temperature in K: 6500
- Scene: 25 (value: **000e0d0000000000000000c803**)

marcosrogers, 2022-08-11:
After further testing, it may be an issue with Localtuya (see [localtuya issue #983](https://github.com/rospogrigio/localtuya/issues/983)) not reading the DPS correctly, but it may also be that the bulb doesn't report its DPS correctly.

Of the three of these bulbs that I have, I did get one of them to briefly connect to Localtuya through its GUI setup process, but then it stopped working. I got it working again by setting its scan interval to 10 seconds (previously not enabled).

Still, neither of the other two bulbs would work. Then I edited the `core.config_entries` file to duplicate the working bulb's configuration and update it with the host, deviceid, and localkey from one of the other bulbs. That bulb is now also working.

The third bulb still does not want to connect using the Home Assistant interface.

Also of note, when I look at the entity values from the bulb that did connect in the core.config_entries file,
 - Color 24 is 000003e803e8 (as in [kwiksilver's comment](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/88#issuecomment-1211457977))
 - Scene 25 is 000e0d0000000000000000c803e8 (as it was for me previously)
 - Color temperature 23 is 0, unlike kwiksilver's or my values.

closed 2022-11-30 by Cossid