tuya-cloudcutter/tuya-cloudcutter #90: WB3L - Calex  E27 A60 RGBCCT 429004.1 v2.0 Light Bulb

ihrapsa, 2022-05-16:
Hi,
This is the dump of the bulb that once used to have an esp chip and be "[tasmotizable](https://templates.blakadder.com/calex-429004.html)". I've dumped it with [bk7231tools](https://github.com/khalednassar/bk7231tools) in it's 'virgin' form as I haven't connected to any tuya app yet.
[dump_calex_rgbcct_bulb.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8702853/dump_calex_rgbcct_bulb.bin.zip)

I've tried a couple of already existing profiles (Calex, Generic-Tuya, LSC) but all of them ended up with an error like this:

```
Using PSK v1 - Received PSK ID version 02
[W 220516 16:53:21 iostream:1406] SSL Error on 12 ('10.42.42.23', 49154): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
```

Let me know if I should dump the bin after I connect it to a tuya server as well.

Thanks!!!! 😃

MoogleTroupe, 2022-05-16:
@khalednassar dissection fails on this bin
`RBL containers:
        0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xe470]
Traceback (most recent call last):
  File "bk7231tools.py", line 279, in <module>
    args.handler(args)
  File "bk7231tools.py", line 189, in dissect_dump_file
    output_directory=output_directory, extract=args.extract, with_rbl=args.rbl)
  File "bk7231tools.py", line 89, in __carve_and_write_rbl_containers
    fsout.write(__decrypt_code_partition(partition, container.payload))
  File "bk7231tools.py", line 51, in __decrypt_code_partition
    return cipher.decrypt(payload, partition.mapped_address)
  File "/root/bk7231tools/bk7231tools/crypto/code.py", line 70, in decrypt
    return self.encrypt(data, stream_start_offset=stream_start_offset)
  File "/root/bk7231tools/bk7231tools/crypto/code.py", line 59, in encrypt
    raise ValueError(f"Given data length {len(data)} is not a multiple of block length {self.BLOCK_LENGTH_BYTES}")
ValueError: Given data length 58480 is not a multiple of block length 32
`

ihrapsa, 2022-05-16:
Older version of bk7231tools (before 22.04.2022) disects it successfully.

```
% python3 bk7231tools.py dissect_dump -e -O dump_extract_dir dump_calex_rgbcct_bulb.bin  
RBL containers:
	0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xe470] - extracted to dump_extract_dir/dump_calex_rgbcct_bulb_bootloader_1.00.bin
	0x129f0a: app - [encoding_algorithm=NONE, size=0xeb560] - extracted to dump_extract_dir/dump_calex_rgbcct_bulb_app_1.00.bin
```
	
	
[image]

ihrapsa, 2022-05-16:
This is the bin dumped after connecting to tuya servers to help with profile building 😃 :
[dump_calex_tuya_server.bin.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8703431/dump_calex_tuya_server.bin.zip)

ihrapsa, 2022-05-17:
> @khalednassar dissection fails on this bin
> `RBL containers: 0x10f9a: bootloader - [encoding_algorithm=NONE, size=0xe470] Traceback (most recent call last): File "bk7231tools.py", line 279, in [...]

Commit [90c8617](https://github.com/khalednassar/bk7231tools/commit/90c861785b9b4b880e2a2fd33d816d8c66ffda29) fixed this. Dissecting with decrypting proceeds successfully now.

ihrapsa, 2022-06-18:
Thanks a lot for the profile. It works! 🙂

closed 2022-08-04 by ihrapsa