tuya-cloudcutter/tuya-cloudcutter #92: E14 generic Smart Life bulb! Photos, again new module,  [WBL2_M1][BK7231N]

openshwprojects, 2022-05-18:
Another bulb that 
[image]
[image]
[image]
[image]
[image]
[image]
really needs OTA hack, because TX and RX is covered by PCB and you have to otherwise desolder module with hot AIR!
[image]
[image]
And the dump:
[bk7231s_dump-2022- 5- 6.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8713629/bk7231s_dump-2022-.5-.6.zip)

khalednassar, 2022-06-10:
@openshwprojects are you sure that's the correct dump? It matches the Gosund BK7231T from #85. It's definitely not a dump from a BK7231N chip.

Cossid, 2022-12-01:
Confirming the attached dump has the same uuid as #85 so this is an invalid dump.  All 4 are the same as #85.  Closing, if you want it re-opened, please attach a corrected dump.

closed 2022-12-01 by Cossid