tuya-cloudcutter/tuya-cloudcutter #94: WB2L - Aldi/Lightway SMART-AL2017-TGTS Downlight

f3sty, 2022-05-22:
Lightway 8W Smart LED Downlight model SMART-AL2017-TGTS, Sold by Aldi Australia

[image]

[image]

Dumped in its virgin state with bk7231tools
[SMART-AL2017-TGTS.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8749010/SMART-AL2017-TGTS.zip)

openshwprojects, 2022-05-23:
Can you also join to a dummy wifi network, configure with app and attach post-join dump?

pastcompute, 2022-06-05:
Hi there, I have one of these, I'm not presently in a position to try dumping it but I was able to run cloudcutter using the tuya generic device option, and it seems to run the exploit as it updated the SSID to the cloudcutter device and it connects back after power cycle.
However the tool is then failing after the device connects back to the cloudcutter wifi ssid, with a "SSL decrypt or bad MAC error" and from some digging it seems that the TLS PSK or algorithm must be mismatched, either that or I've misunderstood how this is working

(I will try and post a log tomorrow)

khalednassar, 2022-06-10:
Profile is added. Can someone give it a try?

Interestingly, the app code matches a bunch of LSC devices, which do already work:
```
970716 Smart LED Tunable White E27 806 Lumen
970700 Smart LED Tunable White GU10 345 Lumen
970710 Smart LED RGB + Tunable White GU10 380 Lumen
970820 Smart LED Downlight Tunable White 360 Lumen
970724 Smart LED RGB + Tunable White E27 806 Lumen
```
Do wonder if there are any differences for the device @pastcompute is trying on, though.

pastcompute, 2022-06-13:
This seems to have worked for me now - I can observe the device on my wifi, with some caveats:
- I had to patch the scripts, for some reason the `service NetworkManager stop` (etc) commands don't find `service`, and I had to change it to `systemctl stop NetworkManager` (etc) instead - I'm guessing this is due to whatever version of Debian is running on my nuc
- the hostname shows up through DHCP to my Wifi AP as `wlan0` which is a bit odd

Now I'm just having trouble getting tinytuya to talk to it but I guess thats a different problem. It is broadcasting something on port 6667, and from Wireshark I can see tinytuya opening a connection on port 6668 but very few bytes seem to be exchanged, I'm possibly just getting the keys wrong somehow

f3sty, 2022-06-18:
> Profile is added. Can someone give it a try?
That profile worked for me, thanks. 
Power, dimmer, colourtemp and mode switching all working as expected via tinytuya.

khalednassar, 2022-06-21:
> This seems to have worked for me now - I can observe the device on my wifi, with some caveats:
>
>     * I had to patch the scripts, for some reason the `service NetworkManager stop` (etc) commands [...]
Might be worth switching to the systemd variant in general. I'll take a look at that, thanks.

> 
>     * the hostname shows up through DHCP to my Wifi AP as `wlan0` which is a bit odd
> 
I'm afraid that's just how the tuya firmware acts, can't particularly change it without changing the firmware itself. They do that in completely stock operation when onboarded via the apps as well.

> Now I'm just having trouble getting tinytuya to talk to it but I guess thats a different problem. It is broadcasting something on port 6667, and from Wireshark I can see tinytuya opening a [...]

The code template below works, note the `updatedps` call which is necessary to "refresh" the device in many cases
```python
import tinytuya

d = tinytuya.BulbDevice('<deviceID>', '<deviceIP>', '<deviceLocalKey>')
d.version = 3.3 # hack to set the version without communicating with the device
d.updatedps()
d.set_version(3.3)
# Do whatever
```

chris-chensy, 2022-10-09:
The profile worked. And tinytuya works fine. However, this light also originally supports colour change but it seems this feature is missing in the profile. Therefore, tinytuya was unable to change the colour of the light. Can anyone add this feature? I suppose it's datapoint 24 (?).

closed 2022-11-30 by Cossid