tuya-cloudcutter/tuya-cloudcutter #95: FLSNT BR30 WB2L BK7231T

Cossid, 2022-05-23:
Product: FLSNT BR30 RGBWCT ( https://smile.amazon.com/dp/B086GZQM1J )
Module: WB2L
Chip: BK7231T
Flash Size: 2M
New AP Prefix: "SL-FLSNT-"
Product Image:
[image]
Attached [FLSNT_BR30.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8751129/FLSNT_BR30.zip):
- Untouched OEM bin dump: oem_flsnt2m.bin
- SmartLife Joined (throw-away AP, SSID GUEST-SL) bin dump: oem_flsnt2m_sl_joined.bin

Needs verification and exploit

Cossid, 2022-05-23:
haxomatic output:

```
[!] Loading and disassembling code - may take a moment
[+] Code loaded!
[!] Searching for post-vuln code patterns
[+] Found a post-vuln code pattern match!
[+] Matched instructions:
        0xa8318: adds r4, #0xfc
        0xa831a: ldr r3, [r4, #0x50]
        0xa831c: ldr r1, [pc, #0xb0]
        0xa831e: adds r0, r6, #0
[+] Identified lan object register as r4
[!] Searching for JSON object register
[+] Identified JSON object register as r7
[!] Searching for ty_cJSON_Parse function address
[+] ty_cJSON_Parse address: 0xb6b58
[!] Searching for mf_cmd_process gadget address
[+] mf_cmd_process gadget address (THUMB): 0x9f091
[!] Searching for a mov r0, r7 intermediate gadget
[+] Found usable intermediate gadget at address 0xac1f0:
        0xac1f0: ldr r3, [r4]
        0xac1f2: adds r0, r7, #0
        0xac1f4: blx r3
[+] Payload gadgets (THUMB): intermediate_gadget_addr=0xac1f1 mf_cmd_gadget_addr=0x9f091
```

closed 2022-05-24 by Cossid

MoogleTroupe, 2022-05-24:
Also, did you pr use the thumb (ending in 1) or the ending in 0 for the intermediate fixup gadget address? 0 won’t work

MoogleTroupe, 2022-05-24:
Awesome thanks