tuya-cloudcutter/tuya-cloudcutter #96: LSC (Action) 2x5M LED Strip RGBWW

WinSCaP, 2022-05-23:
Product: [LSC Smart Connect digitale ledstrips ](https://www.action.com/nl-nl/p/lsc-smart-connect-digitale-ledstrips/)
New AP Prefix: "SmartLife-"
Product Image: 
[image]
[lsc digital led strips.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8752687/lsc.digital.led.strips.zip)

khalednassar, 2022-06-10:
Is it possible to get the full dump for the device instead of just the dissected segments? The profile will be built but the schema for getting the stock firmware working with the right parameters can't be pulled as it is.

Would also appreciate confirming the article id, it's listed as 3008407 on the linked site. Does that match, and is there perhaps a "secondary" id? (e.g. [this mood light](https://www.action.com/nl-nl/p/lsc-smart-connect-sfeerlamp/) has 3004154 but there are two different "builds" with two different secondary ids, which are 970742 and 970743).

Nonetheless, the exploit payload has been added to #102 which can be merged once the right schema is added to the rest of the profile.

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid

Cossid, 2022-12-21:
A dump from a newer device with 1.0.18 firmware has been made, I back-ported the schema to apply to this older 1.0.14 firmware as well.  A profile is now available.