tuya-cloudcutter/tuya-cloudcutter #100: LSC LED Flood Light with sensor (3003366) - WB3S dump

dkliszewski, 2022-05-31:
[https://templates.blakadder.com/lsc_smart_connect_3003366.html](https://templates.blakadder.com/lsc_smart_connect_3003366.html)
Art. No. 3003366
970740 v1.0
[LSC 3003366 clean dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8803352/LSC.3003366.clean.dump.zip)

Cossid, 2022-10-15:
I'm afraid neither dump attached are complete enough to extract any necessary data.  Please try again with the [bk7231tools](https://github.com/tuya-cloudcutter/bk7231tools)

pcmihnea, 2022-10-22:
What's missing from the dump file?
Is the device by chance similar to one already supported?

Cossid, 2022-11-21:
The app partition is incomplete, and there is not storage partition.  bk7231tools cannot dissect this dump.

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid