tuya-cloudcutter/tuya-cloudcutter #104: Arlec GRID-Connect GLD132HA Dump (WB2L)

JesseMcL, 2022-06-13:
First time trying to use cloudcutter or work with the tuya BK modules.

Here's a few flash dumps I took from this device using bkwriter (and uartreader), tried both in factory state and after joining a test network (with internet access), however must have had something go wrong since the dumps are identical, and I only got the device to register to tuya cloud once, but it never worked through the tuya smart app, and i could also never get it to re-register with tuya, it always times out after joining wifi now and usually with an error about not resolving m2.tuyaeu.com on the wb2l debug console.

Then captured another dump using bk7231tools which successfully decrypted.

Cloudcutter also seemed to not proceed past getting the device to join the cloudcutter network (using the existing arlec profile), and after that the device just keeps broadcasting an A- wifi network, which cloudcutter would find and join, and report 'device registered' with a different GUID each time, but would also never proceed.

Left overnight and now the WB2L is reporting malloc failures via the debug uart.
[arlec-gld132ha.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8887547/arlec-gld132ha.zip)

I also saved some serial debug logs, if that helps...

Cossid, 2022-10-15:
The Arlec ALD092CHA profile should work for this device.

Cossid, 2022-11-30:
An update has been pushed that should include a profile for this device.  Please update your repo and give it a try.  Please see the instructions page ( https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md ) for any command or input changes.

If the new profile does not work, please re-open this ticket with details (including logs) of the failure.
Any feedback is appreciated.

closed 2022-11-30 by Cossid

Majr25, 2024-08-26:
Hi, have just bought a pack of these for a steal (3 pack itself is called GLD132P3HA, but contains three bulbs marked GLD132HA), but haven't had any luck.

I have tried both the GLD132HA and the ALD092CHA profiles. Every time it connects to the bulb, attempts to run the exploit, the bulb restarts itself (straight back in to AP mode interestingly) then when the script reconnects to it it says the profile did not result in a successful exploit.

I have connected to it with the Tuya app, and the version it reports itself as is 2.9.16. Perhaps this is a different version and something has changed?
I then wiped it, and was able to dump the firmware with Lightleak, which I have attached. So it guess it is still exploitable? [gld132ha.zip](https://github.com/user-attachments/files/16747634/gld132ha.zip)

Please let me know if there's any more details I can provide!

Cossid, 2024-09-04:
Updated adding a profile for both 2.9.15 and 2.9.16