tuya-cloudcutter/tuya-cloudcutter #107: Merkury Innovations Smart WIFI Dimmable Color + White LED Bulbs (SKU: MI-BW944-999W) (BK7231T WB2L)

mascencerro, 2022-06-19:
Here is firmware and boot log from virgin (unconnected) Smart Bulb from Wal-Mart
https://mygeeni.com/collections/colored-lights/products/merkury-innovations-dimmable-75w-equivalent-wi-fi-smart-bulb-color-4-pack
[MI-BW944-999W.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/8936260/MI-BW944-999W.zip)

mascencerro, 2022-06-20:
I've been made aware that this is only the app dump and not the full firmware. I has suspicion that the boot will be the same as the other bulb I made an issue with full firmware, but will find another device to tear down for a full firmware soon.

Cossid, 2022-10-15:
The Phopollo SBE2606 profile should work with this device.

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid