tuya-cloudcutter/tuya-cloudcutter #111: MK601 CB2S smart switch (a smart switch very hard to program, because the RX/TX are covered by PCB)

openshwprojects, 2022-06-30:
A flash dump from Elektroda user, it's not a full version sadly, but maybe you'll use it anyway

[image]
[image]
[image]
[image]
[image]

[bk7231s_dump-2022- 6-29-23- 4-31.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9016295/bk7231s_dump-2022-.6-29-23-.4-31.zip)

openshwprojects, 2022-06-30:
this one might be better
[bk7231s_dump-2022- 6-29-23-16-45.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9019088/bk7231s_dump-2022-.6-29-23-16-45.zip)

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid