tuya-cloudcutter/tuya-cloudcutter #113: LEDVANCE Wifi SMART+ Planon Frameless LED RGBW 30x30 3000-6500K

Xaymar, 2022-07-09:
Trying to get [this "lamp"](https://shop.ledvance.com/products/ledvance-smart-multicolor-300x300) offline into a localtuya/minituya. The existing LEDVANCE exploit works up until the point where it attempts to query `h3.iot-dns.com` and fails with a generic SSL Error.

Since I plan on moving these lamps to not-important rooms anyway, is there any way I can help with getting these exploited? I can probably disassemble one of them for a dump, assuming I can find the necessary dumping tools in stock anywhere.

Cossid, 2022-10-15:
If you have the device already connected to Smart Life, can you provide the version listed under "device update" in the Smart Life app?

Xaymar, 2022-11-06:
Sorry for the late reply, I ended up throwing these away due to an automatic firmware update turning red into purple, purple into purple, blue into yellow, and yellow into green. I can dig them out of the e-waste pile, though i probably won't be able to connect them to proper 230V power.

Cossid, 2022-12-10:
Closing as stale.  Re-open if you decide to try the devices again.

closed 2022-12-10 by Cossid