tuya-cloudcutter/tuya-cloudcutter #115: SC500W v3 - Curtain / Blind controller

Sineos, 2022-07-12:
Link: [aliexpress.com/item/1005002710349987.html](https://www.aliexpress.com/item/1005002710349987.html)
Module: WB2S
Dump: [firmware.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9095047/firmware.zip)

Cossid, 2022-10-15:
I cannot provide a Tuya schema for stock firmware with the provided dump, but if you plan to use a 3rd party firmware, the following profile should work:

```
{
    "chip": "BK7231T",
    "payload": "eyJhdXprZXkiOiJBVVRIS0VZQUFBQUFBQUFBIiwidXVpZCI6IlVVSURBQUFBQUFBQSIsInBza0tleSI6IiIsInByb2RfdGVzdCI6ZmFsc2UsImFwX3NzaWQiOiJBIiwic3NpZCI6IkEiLCJ0b2tlbiI6IkFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQUFBQTUyCiJ9",
    "authkey_template": "AUTHKEYAAAAAAAAA",
    "uuid_template": "UUIDAAAAAAAA",
    "datagram_padding": "QkJCQqloCQCpaAkAqWgJAKloCQCpaAkAqWgJAKloCQCpaAkAqWgJAA=="
}
```

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid