tuya-cloudcutter/tuya-cloudcutter #117: LSC Moodlight 970743

beaune33, 2022-07-21:
Dear all,

I tried cloudcutter with the mentioned device, which is already in the device list. I tried the device with the script test_device_exploitable.py, which seemed to be successful. Afterwards I set up a raspberry PI 3B and tried cloud cutter. On the first view it looked good, but te device did not stop blinking, and in the end returned without exchanged keys. I add the log. Any idea what I could try?

`./run_detach.sh -r <SSID> <password> wlan0 3004154-970743`
```

Found access point name: "SmartLife-BD7C", trying to connect..
Gerät »wlan0« wurde erfolgreich mit »616c5c68-554c-4653-981a-d60245b03a91« aktiviert.
Wiping NetworkManager configs
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-BD7C", trying to connect..
Gerät »wlan0« wurde erfolgreich mit »93bf6378-e087-44b7-a880-45a347f1ab80« aktiviert.
Wiping NetworkManager configs
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-BD7C", trying to connect..
Gerät »wlan0« wurde erfolgreich mit »ff711d42-4dfc-4d75-b305-d9a4cc86b6bb« aktiviert.
Wiping NetworkManager configs
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-BD7C", trying to connect..
Gerät »wlan0« wurde erfolgreich mit »106d95d4-b8f5-49e4-aa2b-3e68e92a444e« aktiviert.
Wiping NetworkManager configs
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Found access point name: "SmartLife-BD7C", trying to connect..
Gerät »wlan0« wurde erfolgreich mit »cd6a74d9-24cc-4f61-a9e7-54760cce76c7« aktiviert.
Error, could not connect to SSID.
Failed to connect, please run this script again
```

beaune33, 2022-07-21:
I am one step further: The point is that I set the language of my PI to german, so we have to deal with german return values in common.sh:
        `if [ "${AP_STATUS}" != "aktiviert" ]; then`
instead of
       ` if [ "${AP_STATUS}" != "activated" ]; then`
After all, key exchanging works.

Now I would like to flash a custom firmware, and I wonder how to do. 

- I copied my firmware into the folder custom-firmware
- I called `./run_flash.sh –r wlan0 LSC/3004154-970743 OpenBK7231N_UG_1.10.4.bin` in different variations (file name only, path + filename,...). 
- It all ends in different error messages like
    > Firmware OpenBK7231N_UG_1.10.4.bin does not exist or not a file
    > Fehler: Gerät »–r« wurde nicht gefunden.

So, I am not sure what to enter as filename: the filename only or any path. And if a path is necessary: why does the script think the -r could be a device name... ist ist an escaping problem? Or is there any other check done, s that my OpenBeken binary is rejected.

Would be glad if anybody could help to figure out how the script works. 
Thanks in advance!

Cossid, 2022-10-15:
Are you running elevated?  you need to run elevated for the wireless AP stuff to work.

You do not *need* to provide the model and file in the command.  If you don't provide them, you will be presented menus to select them instead, it will give you a list of available options.  Try that method instead?

Cossid, 2022-12-08:
I have switched to checking for the connected AP gateway, which is a static IP instead of a word which can be internationalized.

Other questions have been addressed in more recent commits or are unanswered/stale.  Closing this ticket.  If there are further unresolved issues, please feel free to re-open it.

closed 2022-12-08 by Cossid