tuya-cloudcutter/tuya-cloudcutter #118: VEARGREE ATMS1601

PetervdPerk, 2022-07-24:
Link: https://nl.aliexpress.com/item/32908926241.html?spm=a2g0o.order_list.0.0.11ae79d2zDPALS&gatewayAdapt=glo2nld
Module: WB2S
Dump: [firmware.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9175663/firmware.zip)

PetervdPerk, 2022-07-30:
Any guide to make your own device profile?
These are unusable when you try to open them up, so an OTA mechanism to change the sotware would make this a nice module.

Cossid, 2022-10-15:
The Aoycocr X5P profile should work for this device.

Cossid, 2022-12-09:
This post has been tagged as new-dump-needed and has had not response.  This issue is being closed as stale.  To re-open this issue, please provide a new dump following the guildines in the [device dumps](https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps) section of the Readme.

closed 2022-12-09 by Cossid