tuya-cloudcutter/tuya-cloudcutter #120: Feit Smart bulbs - BPPLUG-SMART-3

harsh2029, 2022-07-29:
This is a famous Costco 3-pack Feit smart plugs.
Extracted flash dumps from 2 of these models having different chips.
The old model has WB2S and new one has CB2S
[Feit_BPPLUG-SMART-3.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9223071/Feit_BPPLUG-SMART-3.zip)
.

Cossid, 2022-10-15:
This is a really peculiar case.  That WB2S is an anomaly.  It is the only T device so far that appears to be compiled against the newer N SDK.  There is no current exploit for that one as there are no other completed N profiles with that same version.

The CB2S version should work with the QIACHIP Smart-Switch profile.

caver01, 2022-12-09:
I made the costco purchase last year but mine say:
Item: BPPLUG/WIFI/3
Cracked one open and confirmed it has WB2S.

I tried to flash OTA using Tuya-Cloudcutter (after a successful session flashing 8 bulbs-THANK YOU!)
Got as far firmware request coming in and hit the SSL Decryption failed error.
Tried all 1.1.2 firmware choices available (various switches from different manufacturers) since I ma trying to replace firware with OpenBK. All the same.

Is there anything I should try? With WB2S I should be flashing with the T version of OpenBK, but can’t get past the SSL. Am I out of luck?

Cossid, 2022-12-09:
Yeah, the WB2S with firmware 1.1.2 are different, they're why this issue is marked with `it's complicated`.  We know the device is vulnerable, but it needs some reverse engineering to find the correct offsets, as they are different pattern than the usual T or N devices.  Unfortunately, all you can do at this point is wait for this device.  This issue will be updated when there is progress on it.

fifa255, 2023-01-09:
would firmware1.0.7 on both the Main and MCU module on my WB2S Feit plug work with tuyacloudcutter

Cossid, 2023-01-10:
We finally got this straightened out, and there is now a profile for both the T and N versions of this plug.

Cossid, 2023-01-10:
> would firmware1.0.7 on both the Main and MCU module on my WB2S Feit plug work with tuyacloudcutter

@fifa255 are you sure it's 1.0.7?  We have 1.1.2 for T and 1.1.7 for N.  We would need a dump for 1.0.7, which you could post as a separate issue if you're able.

closed 2023-01-10 by Cossid

caver01, 2023-01-11:
> We finally got this straightened out, and there is now a profile for both the T and N versions of this plug.

This is fantastic! Much appreciated. Will give mine a try. Is it important that I post confirmation that it worked? I suppose I need to update my clone of the repo first. . .

Cossid, 2023-01-11:
Yes, you will need the latest changes for the BK7231T (WB2S) model for sure.  At this point we have 2 confirmations on BK7231T devices with this less common 2.3.2 SDK, but additional confirmations couldn't hurt.

caver01, 2023-01-11:
Confirmed. Using a RPi host, I successfully flashed my Feit plug with BK7231T OpenBeken firmware. Got it up and running on my local network, but now at a loss for configuring GPIO template. Still, happy that I was able to exploit and flash firmware.

Tried the second plug, but it would not detect the plug’s AP (but I could see it was there using my ipad). Rebooted my raspberry P hosti to try again, but it would not start Dockerd. Docker seems to be totally screwed up now. Not sure what happened there, but going to try starting from a fresh image and rebuilding everything from scratch.

But at least one worked, so I know it is possible with a working cloudcutter host.

If anyone has a suggestion for setting the relay,switch and LED on these plugs using OpenBK I’d appreciate it.

caver01, 2023-01-11:
Just for reference, for my Feit plugs, the following config works for OpenBK (Tasmota-like setup):
P10 (RXD1)=LED_n,0
P24 (PWM4)=Rel,0
P26 (PWM5)=Btn,0,0

caver01, 2023-01-13:
Tried the next 2 plugs and the process behaves as though it worked, but it is not really running the new firmware at all. Here is what looks like a successful session, bur results in a device to which I cannot really use, though I can get it to go into WAP mode and it shows up as A-1234 or something. Not the firmware I expected. What went wrong? I have two plugs that are both behaving this way now.

`Using PSK v1 - Received PSK ID version 01
[LOG (Client)] Request: HTTPServerRequest(protocol='https', host='h3.iot-dns.com', method='POST', uri='/v2/url_config', version='HTTP/1.1', remote_ip='10.42.42.21')
[LOG (Client)] ==== Request body ===
b'{"region":"AA","env":"","uuid":"I8ShwwmMVHho","ts":93,"sign":"1GMBqsw76dsP56ySRRNbcrIBPsa4UNc0rWgS/n0KBo0=","config":[{"key":"httpUrl","need_ca":false},{"key":"mqttUrl","need_ca":false},{"key":"httpsPSKUrl","need_ca":false},{"key":"mqttsPSKUrl","need_ca":false}]}'
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {"caArr":null,"httpUrl":{"addr":"http://10.42.42.1/d.json","ips":["10.42.42.1"]},"mqttUrl":{"addr":"10.42.42.1:1883","ips":["10.42.42.1"]},"ttl":600}
[I 230113 03:59:21 web:2271] 200 POST /v2/url_config (10.42.42.21) 114.06ms
[LOG (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.active&et=1&t=95&uuid=I8ShwwmMVHho&v=4.4&sign=7ecb8b572d8221e801e1e1ce306c25d8', version='HTTP/1.1', remote_ip='10.42.42.21')
[LOG (Client)] ==== Request body ===
{'token': 'AAAAAA', 'softVer': '1.1.2', 'productKey': 'key9y7ndhaqtyp5g', 'protocolVer': '2.2', 'baselineVer': '40.00', 'productKeyStr': 'key9y7ndhaqtyp5g', 'hid': '7cf6666448fa', 'modules': '[{"otaChannel":9,"softVer":"1.1.2","online":true}]', 'devAttribute': 611, 'cadVer': '1.0.4', 'cdVer': '1.0.0', 'options': '{"isFK":true,"otaChannel":0}', 't': 95}
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {'result': {'schema': '[{"type":"obj","mode":"rw","property":{"type":"bool"},"id":1},{"type":"obj","mode":"rw","property":{"min":0,"max":86400,"scale":0,"step":1,"type":"value"},"id":9},{"type":"obj","mode":"rw","property":{"range":["off","on","memory"],"type":"enum"},"id":38}]', 'devId': 'ZMNKeLCpk1gEnW1MSIyp', 'resetFactory': False, 'timeZone': '+02:00', 'capability': 1025, 'secKey': 'LxGFm8He0wIVUe4h', 'stdTimeZone': '+01:00', 'schemaId': '00000335js', 'dstIntervals': [], 'localKey': '65XOB16ad8xWASZV'}, 'success': True, 't': 1673582361}
[I 230113 03:59:21 web:2271] 200 POST /d.json?a=tuya.device.active&et=1&t=95&uuid=I8ShwwmMVHho&v=4.4&sign=7ecb8b572d8221e801e1e1ce306c25d8 (10.42.42.21) 14.16ms
[LOG (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.dynamic.config.get&devId=ZMNKeLCpk1gEnW1MSIyp&et=1&t=1673582361&v=2.0&sign=6dbc467eaf8d972a13487ece34fff158', version='HTTP/1.1', remote_ip='10.42.42.21')
[LOG (Client)] ==== Request body ===
{'type': '["timezone"]', 't': 1673582361}
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {'result': {'ackId': '0-0', 'validTime': 1800, 'time': 1673582363, 'config': {'stdTimeZone': '+01:00', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200]]}, 'timezone': {'ackId': '0-0', 'validTime': 1800, 'time': 1673582363, 'config': {'stdTimeZone': '+01:00', 'dstIntervals': [[1648342800, 1667091600], [1679792400, 1698541200]]}}}, 't': 1673582363, 'success': True}
[I 230113 03:59:23 web:2271] 200 POST /d.json?a=tuya.device.dynamic.config.get&devId=ZMNKeLCpk1gEnW1MSIyp&et=1&t=1673582361&v=2.0&sign=6dbc467eaf8d972a13487ece34fff158 (10.42.42.21) 16.28ms
[LOG (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.mesh.get&devId=ZMNKeLCpk1gEnW1MSIyp&et=1&t=1673582361&v=1.0&sign=1a34734b854a2db6e5d2818d2e98cf89', version='HTTP/1.1', remote_ip='10.42.42.21')
[LOG (Client)] ==== Request body ===
{'t': 1673582361}
[LOG (Client)] ==== End request body ===
!!! Endpoint default response not found - tuya.device.mesh.get
[LOG (Server)] Response:  {'success': True, 't': 1673582363}
[I 230113 03:59:23 web:2271] 200 POST /d.json?a=tuya.device.mesh.get&devId=ZMNKeLCpk1gEnW1MSIyp&et=1&t=1673582361&v=1.0&sign=1a34734b854a2db6e5d2818d2e98cf89 (10.42.42.21) 9.75ms
[LOG (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.uuid.pskkey.get&et=1&t=1673582365&uuid=I8ShwwmMVHho&v=1.0&sign=86ddfd75716cfef47e790811249b2aeb', version='HTTP/1.1', remote_ip='10.42.42.21')
[LOG (Client)] ==== Request body ===
{'t': 1673582365}
[LOG (Client)] ==== End request body ===
[LOG (Server)] Response:  {'result': {'pskKey': ''}, 't': 1673582366, 'success': True}
[I 230113 03:59:26 web:2271] 200 POST /d.json?a=tuya.device.uuid.pskkey.get&et=1&t=1673582365&uuid=I8ShwwmMVHho&v=1.0&sign=86ddfd75716cfef47e790811249b2aeb (10.42.42.21) 108.53ms
Sending firmware update message {"data":{"firmwareType":0},"protocol":15,"t":1673582368} using protocol 2.2
Firmware update messages triggered. Device will download and reset. Exiting in 30 seconds.
[LOG (Client)] Request: HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.upgrade.get&devId=ZMNKeLCpk1gEnW1MSIyp&et=1&t=1673582367&v=4.4&sign=00f9958ebe38b559dab9c0eba478ea88', version='HTTP/1.1', remote_ip='10.42.42.21')
[LOG (Client)] ==== Request body ===
{'type': 0, 't': 1673582367}
[LOG (Client)] ==== End request body ===
[... cut ...]
Jan 13 03:59:54 dnsmasq-dhcp[15]: 502144481 sent size:  4 option: 58 T1  6h
Jan 13 03:59:54 dnsmasq-dhcp[15]: 502144481 sent size:  4 option: 59 T2  10h30m
Jan 13 03:59:54 dnsmasq-dhcp[15]: 502144481 sent size:  4 option:  1 netmask  255.255.255.0
Jan 13 03:59:54 dnsmasq-dhcp[15]: 502144481 sent size:  4 option: 28 broadcast  10.42.42.255
Jan 13 03:59:54 dnsmasq-dhcp[15]: 502144481 sent size:  4 option:  3 router  10.42.42.1
Jan 13 03:59:54 dnsmasq-dhcp[15]: 502144481 sent size:  4 option:  6 dns-server  10.42.42.1
pi@cloudcutter:~/tuya-cloudcutter $ 
`

Cossid, 2023-01-13:
That just means the firmware update didn't take.  It happens sometimes, and there is no way for cloudcutter to give a better message.  You can just repeat the process, and eventually it will work.  As always, make sure your target and cutter devices are as close to each other as possible.

caver01, 2023-01-13:
Thanks. I tried a few more times and it finally worked on one of them. I will try the third tomorrow. I cannot get this second one to connect to my Wi-Fi, but it did flash and I can get to the configuration in safe mode. Thanks for all of the replies here!

caver01, 2023-01-15:
Got them both flashed and working great! For anyone reading this after the fact, it definitely helps to bring the device closer to your OTA cloudcutter host. In my case, a Raspberry Pi. Placing them within a few feet of each other, I was able to run the exploit easily.

@Cossid thanks for all of the personal touches here. I am considering picking up inexpensive dimmer switches I saw today at costco and giving them a try. Very exciting!