tuya-cloudcutter/tuya-cloudcutter #121: Feit Smart Bulbs - BPA8000RGBW2P

harsh2029, 2022-07-29:
This dump is for 2-pack Feit Smart bulbs from Costco using WB2L module
[Feit_BPA800RGBW2.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9223131/Feit_BPA800RGBW2.zip)
.

Cossid, 2022-10-15:
The Phopollo SBE2606 profile should work for this bulb.

rickyelopez, 2022-10-16:
Just tested with Phopollo SBE2606 profile and it didn't work. Got stuck at the point where the device is supposed to connect to the cloudcutter network.

```
Found access point name: "A-6F4F", trying to connect..
Device 'device' successfully activated...
Connected to access point.
Configured device to connect to 'cloudcutterflash'
Device is connecting to 'cloudcutterflash' access point. Passphrase for the AP is 'abcdabcd' (without ')
Flashing custom firmware ..
==> Wait for 20-30 seconds for the device to connect to 'cloudcutterflash'. This script will then show the firmware upgrade requests sent by the device.
Using WLAN adapter: device
Oct 16 23:04:12 dnsmasq[15]: started, version 2.80 cachesize 150
Oct 16 23:04:12 dnsmasq[15]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Oct 16 23:04:12 dnsmasq-dhcp[15]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
Oct 16 23:04:12 dnsmasq-dhcp[15]: DHCP, sockets bound exclusively to interface wlp0s20f3
Oct 16 23:04:12 dnsmasq[15]: read /etc/hosts - 2 addresses
Configuration file: /dev/stdin
Using interface device with hwaddr aa:bb:cc:dd:ee:ff and ssid "cloudcutterflash"
wlp0s20f3: interface state UNINITIALIZED->ENABLED
wlp0s20f3: AP-ENABLED
```

just hangs there, waited a few minutes and no dice.

Cossid, 2022-10-16:
The fact you got an AP prefix of "A-" means the profile is the right one.  Cloudcutter isn't perfect and sometime you have to retry multiple times, especially if it has failed previously.  I'd highly recommend restarting the device running cloudcutter anytime there is an issue that appears to be lack of network connection.  Also make sure to keep the target device as close to the device running cloudcutter as possible.

rickyelopez, 2022-10-16:
Hmm ok, I've restarted the laptop I'm trying this on and have also turned off the bulb itself and tried again, but still no luck. Before I saw this issue, I attempted the process using the Feit/OM60-RGBW-CA-AG profile. Could that have had an impact? Do you have any other suggestions for me to try?

tony-fav, 2022-10-16:
Once it's an A- access point, you can use the Tuya app to connect it to the cloudcutter AP

rickyelopez, 2022-10-17:
Oh I see, maybe I misunderstood the instructions? I thought it would connect to the cloudcutter AP on its own? Anyway, I tried using the Tuya app to get it to connect, but it didn't work. Tried both slow blinking and fast blinking, neither works. It's clear the app is communicating with the bulb because as soon as I hit the "start pairing" button, it stops blinking. But nothing connects to the cloudcutter AP and the script doesn't print anything.

tony-fav, 2022-10-17:
The app trick is just a way to give the device another try to connect to the cloudcutter AP.

Were there log messages when you connected your phone to cloudcutter AP? There should have been.

Cossid, 2022-10-17:
Your assumption is correct, it should connect to the AP automatically.  For cloudcutter, you should always be using slow blinking mode.

In the instruction where it says to cycle the power after running the initial exploit, try leaving the device sit for a few minutes without power cycling, it should reboot on its own and restart in the correct mode.

It could be there are other conflicting required services causing issues as well, I forget what besides mqtt (dnsmasq maybe?).  Before you run anything, if you have tuya-convert installed start that up once (it has a nice script for killing necessary services), then cancel out of that and try cloudcutter.

rickyelopez, 2022-10-17:
Ok, by using the tuya-convert start script I figured out that dnsmasq was running. After killing that and trying cloudcutter again, I was able to get the bulb to connect to the AP using the Tuya app, but then I hit this:

```
Using PSK v1 - Received PSK ID version 01
[I 221017 00:54:57 web:2239] 200 POST /device/url_config (10.42.42.35) 47.38ms
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.AUTHKEY. Retrying with other key choice
Decryption failed due to padding error with key choice TuyaCipherKeyChoice.SECKEY. Retrying is false - bailing out
[E 221017 00:54:58 web:1789] Uncaught exception POST /d.json?a=tuya.device.active&et=1&t=69&uuid=M5p3tCNEi4kp&v=4.4&sign=832a78477e40ba9356466ab9ea398ef8 (10.42.42.35)
    HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/d.json?a=tuya.device.active&et=1&t=69&uuid=M5p3tCNEi4kp&v=4.4&sign=832a78477e40ba9356466ab9ea398ef8', version='HTTP/1.1', remote_ip='10.42.42.35')
    Traceback (most recent call last):
      File "/src/cloudcutter/crypto/tuyacipher.py", line 51, in __decrypt_with_retry
        return padding.unpad(decrypted, block_size=16)
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "/src/.venv/lib/python3.9/site-packages/tornado/web.py", line 1702, in _execute
        result = method(*self.path_args, **self.path_kwargs)
      File "/src/cloudcutter/protocol/handlers.py", line 81, in post
        request_body = self.__decrypt_request_body(key_choice)
      File "/src/cloudcutter/protocol/handlers.py", line 129, in __decrypt_request_body
        decrypted = self.cipher.decrypt(body, key_choice).decode('utf-8')
      File "/src/cloudcutter/crypto/tuyacipher.py", line 30, in decrypt
        return self.__decrypt_with_retry(data, key_choice)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 59, in __decrypt_with_retry
        return self.__decrypt_with_retry(data, key_choice, False)
      File "/src/cloudcutter/crypto/tuyacipher.py", line 56, in __decrypt_with_retry
        raise e
      File "/src/cloudcutter/crypto/tuyacipher.py", line 51, in __decrypt_with_retry
        return padding.unpad(decrypted, block_size=16)
      File "/src/.venv/lib/python3.9/site-packages/Cryptodome/Util/Padding.py", line 92, in unpad
        raise ValueError("Padding is incorrect.")
    ValueError: Padding is incorrect.
[E 221017 00:54:58 web:2239] 500 POST /d.json?a=tuya.device.active&et=1&t=69&uuid=M5p3tCNEi4kp&v=4.4&sign=832a78477e40ba9356466ab9ea398ef8 (10.42.42.35) 13.80ms
```

This loops a bunch of times, and then everything just hangs.

Cossid, 2022-10-17:
Don't use the Tuya App, I'm not sure that is supported at all, it looks like it is likely making requests that cloudcutter doesn't have scripts for.

rickyelopez, 2022-10-17:
I only used the app after the bulb did not connect to the AP for a few minutes. Is there anything else I can do to get it to connect?

Cossid, 2022-10-17:
Just start the process over from the beginning and give it a little time.  I've never had a device not at least connect after getting an A- AP prefix, but some devices took a few tries going through the process to full finish.

rickyelopez, 2022-10-17:
Ok cool looks like this time around it made it all the way on its own, except it still ends up looping the error I posted previously. Didn't use the Tuya app and it connected to the AP, though, so I guess that's progress?

Cossid, 2022-10-17:
I can't speak to what that error is, I have never seen it before.  You might consider trying to pair the device up with Smart Life, then starting the cloud cutter process over.

rickyelopez, 2022-10-17:
Huh interesting, looks like I can't even do that... Seems like something messed up this bulb. Before I take it apart and try to flash it the hard way, any suggestions for a factory reset or something?

Cossid, 2022-10-17:
Not that I can think of, I think putting it into slow blinking mode pretty much is the reset.  The only other option is if it is paired to SmartLife, you can "remove + wipe data" but there is not option to do that when it's not paired.

rickyelopez, 2022-10-17:
rip alright well guess I'll have to do that. Interestingly, tried it on the other bulb from the two-pack and it worked perfectly first try. Wonder what went wrong with the first one...

Cossid, 2022-11-30:
An update has been pushed that should include a profile for this device.  Please update your repo and give it a try.  Please see the instructions page ( https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/INSTRUCTIONS.md ) for any command or input changes.

If the new profile does not work, please re-open this ticket with details (including logs) of the failure.
Any feedback is appreciated.

closed 2022-11-30 by Cossid

BSkando, 2023-03-19:
I have 4 of these bulbs and I am unable to flash any of them with the profile provided.  Any help would be amazing.  Frustrating that I know these bulbs are exploitable.  Here are the logs

`Loading options, please wait...
[?] How do you want to choose the device?: By manufacturer/device name
 > By manufacturer/device name
   By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the brand of your device: Feit
   Enbrighten
   Enshine
   Euarne
   Eyesound
   FCmila
   FLSNT
 > Feit
   Fitop
   Globe Electric
   Globo Lighting
   Gosund
   Hama
   Helloify

[?] Select the article number of your device: BPA800/RGBW/AG/2 RGBCT Bulb
 > BPA800/RGBW/AG/2 RGBCT Bulb
   OM60/RGBW/CA/AG A19 RGBW Bulb
   OM60/RGBW/CA/AG/3 Bulb
   SYW PLUGWIFIG2 Smart Plug v1.0.7
   SYW PLUGWIFIG2 Smart Plug v1.1.0
   SYW PLUGWIFIG2 Smart Plug v1.1.2
   SYW PLUGWIFIG2P Smart Plug
   TAPE16-RGBW-RP LED Strip

[?] Select the firmware version and name: 2.9.16 - BK7231T / oem_bk7231s_light_ty
 > 2.9.16 - BK7231T / oem_bk7231s_light_ty

Selected Device Slug: feit-bpa8000rgbw2p-rgbct-bulb
Selected Profile: oem-bk7231s-light-ty-2.9.16-sdk-1.0.8-40.00

================================================================================
Place your device in AP (slow blink) mode.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "SmartLife-5A84", trying to connect...
Device 'wlan0' successfully activated with 'fe9e798e-fd8a-45f2-b98b-b889002ec435'.
Connected to access point.
Waiting 1 sec to allow device to set itself up...
Running initial exploit toolchain...
Exploit run, saved device config too!
output=/work/configured-devices/Tc2yHQAs3Bm6.deviceconfig
Saved device config in /work/configured-devices/Tc2yHQAs3Bm6.deviceconfig

================================================================================
Power cycle and place your device in AP (slow blink) mode again.  This can usually be accomplished by either:
Power cycling off/on - 3 times and wait for the device to fast-blink, then repeat 3 more times.  Some devices need 4 or 5 times on each side of the pause
Long press the power/reset button on the device until it starts fast-blinking, then releasing, and then holding the power/reset button again until the device starts slow-blinking.
See https://support.tuya.com/en/help/_detail/K9hut3w10nby8 for more information.
================================================================================

Scanning for open Tuya SmartLife AP
.
Found access point name: "SmartLife-5A84", trying to connect...
Error: Connection activation failed: (53) The Wi-Fi network could not be found.
..
Found access point name: "SmartLife-5A84", trying to connect...
Device 'wlan0' successfully activated with 'fe9e798e-fd8a-45f2-b98b-b889002ec435'.
Connected to access point.
================================================================================
[!] The profile you selected did not result in a successful exploit.
================================================================================
`

Cossid, 2023-03-19:
@BSkando That means your device has a different firmware version.  You'll need to start by finding out what firmware version your device has: https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#how-do-i-find-out-what-firmware-version-my-device-has

BSkando, 2023-03-19:
Thank you for the quick response.  I did that before and figured out it is version 1.1.1.  I did try flashing based on version and got the same error.

```
 > By firmware version and name
   From device-profiles (i.e. custom profile)

[?] Select the firmware version and name: 1.1.1 - BK7231T / HYS_03_018_bk7231t_a4y9plclmzpct6ol
   1.1.0 - BK7231N / oem_bk7231n_curtain_switch
   1.1.0 - BK7231T / HYS_03_018_bk7231t_qi4hvlw3xndxepsq
   1.1.0 - BK7231T / HYS_03_018_bk7231t_tjsama0rocl95kdn
   1.1.0 - BK7231T / oem_bk7231s_curtain_switch_nuohua
   1.1.0 - BK7231T / oem_bk7231s_light5_sleepwake_random_globe
   1.1.0 - BK7231T / oem_bk7231s_rnd_switch
 > 1.1.1 - BK7231T / HYS_03_018_bk7231t_a4y9plclmzpct6ol
   1.1.11 - BK7231N / oem_bk7231n_temp_hum_sensor
   1.1.17 - BK7231N / oem_bk7231n_temp_hum_sensor
   1.1.2 - BK7231N / oem_bk7231n_control_switch
   1.1.2 - BK7231T / oem_bk7231s_control_switch
   1.1.2 - BK7231T / oem_bk7231s_hailai_1_plug
   1.1.2 - BK7231T / oem_bk7231s_light_db

[?] Select the brand of your device: Tuya Generic
 > Tuya Generic

[?] Select the article number of your device: MSL2 Fairy Lights 10m 100LED
 > MSL2 Fairy Lights 10m 100LED

Selected Device Slug: tuya-generic-msl2-fairy-lights
Selected Profile: hys-03-018-bk7231t-a4y9plclmzpct6ol-1.1.1-sdk-1.0.2-40.00

```

Cossid, 2023-03-19:
Unfortunately, there is no 1.1.1 for a light bulb (it's not only the version number that matters, that's just the first identifier)
This would require a firmware dump from one device.  https://github.com/tuya-cloudcutter/tuya-cloudcutter#device-dumps has instructions on firmware dumping options.

There is even a possibility that you have the older ESP-based bulbs, which CloudCutter would not work for.  https://github.com/tuya-cloudcutter/tuya-cloudcutter/wiki/FAQ#im-not-sure-what-module-is-in-my-device-how-do-i-find-out has instructions to help narrow that down.

BSkando, 2023-03-19:
I decided to try the other bulb I had from the same package.  I could not see the version number, because the Feit app wanted me to update to 1.1.1.  I tried this profile and the exploit worked:

```
Selected Device Slug: feit-om60-a19-rgbw-bulb
Selected Profile: oem-bk7231s-light-ty-olddp-1.0.3-sdk-2.0.0-30.05
```

Does this mean that I'm out of luck on these other bulbs that are already at 1.1.1?

Cossid, 2023-03-19:
No, probably not.  We have the 1.0.3 firmware, and I can see that I can upgrade that device, so I'll try pulling that firmware and see if it is still exploitable, and if so I'll post a new profile.

Just to be clear, what is the model printed on the device you have that has 1.1.1?

BSkando, 2023-03-19:
Excellent, thank you!

BPA800/RGBW/AG/2(P)

Cossid, 2023-03-19:
Okay, I can't add a profile for that specific device, but I have added one for the OM60 model, but you should be able to to use that bulb's 1.1.1 profile for flashing.  It will probably work correctly only for cutting, but I can't verify/guarantee that.

BSkando, 2023-03-19:
That worked!  Thank you so much for updating that profile!