tuya-cloudcutter/tuya-cloudcutter #125: SSL Error while trying to cut from cloud

beaune33, 2022-08-04:
I managed to flash a custom formware on s Smart Connect Moodlight 970743 sucessfully. Now I tried a second one - without success. Everything works fine, even the cloudcutterflash AP appears. But after

```
Aug  4 13:34:24 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.35
Aug  4 13:34:24 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
```

the following message appears several times:

```
Using PSK v1 - Received PSK ID version 02
[W 220804 13:18:21 iostream:1406] SSL Error on 12 ('10.42.42.35', 50857): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)]
```

After that the LED is glowing static, and nothing more happenes.

Any idea how I could proceed? Maybe too young firmware? Any chance to get back to an older one?

pretoriano80, 2022-08-26:
Same issue here.Gave up on this and using tuya-local integration with Homeassistant .
Also disabled internet access and DNS queries for these devices ,so they are running completely local with Homeassistant + Tuya-local integration .

pretoriano80, 2022-08-26:
@beaune33 had another try and this time i used **FLSNT BR30 WB2L** profile....and it worked just fine,no more SSL errors,

beaune33, 2022-08-29:
@pretoriano80: I gave it a try, but unfortunately no success, same SSL error. Did you really manage to flash  Smart Connect Moodlight 970743 by using FLSNT BR30 WB2L profile? Why did you choose this special profile?

pretoriano80, 2022-08-29:
Mine is a different brand (https://www.ngs.eu/en/smart-home/lighting/gleam-1027c/SMARTHOME/SMT-ILLU-0015/ ) and it has WB2L module (BK7231T).
However ,i was getting the same errors while trying different profiles ,then i saw a discussion about FLSNT BR30 and decided to try that profile too (and it worked ).

But the quest it's still not over for me  because i flashed OpenBeken firmware which does not work good with this device -> https://github.com/openshwprojects/OpenBK7231T_App/issues/150

brunohoff, 2022-09-12:
@beaune33 were you able to flash your device? I also tried the FLSNT without success. 

I`m facing the same issue with a Moes Wall Switch that has WB2L module (BK7231T). For what I looked into the code, the psk variable is not fulfilled.

beaune33, 2022-09-13:
@brunohoff Unfortunately not, but I did not really expect either. The FLSNT contains another module than the LSC Moodlight. I also tried some other profles, but without no success.

In general I wonder, if it still makes sense to follow cloud cutter project. Are there still devices on the market, where I could use it? Or does it only work for old devices? I hoped that it could work for my second device, which I bought in august, because the test_device_exploitable.py succeeded. But as you know I got the SSL error. My first device which I bought last year worked fine, but that's old stuff. Probably I will have to open and flash by wire, as activities in this project seem to starve, sadly

brunohoff, 2022-09-13:
@beaune33, for me, is the exact opposite. I have newer devices with WB3L that I was able to flash with the tuya-cloudcutter method and there is this MOES Wall Switch with the old WB2L chip that I coulnd flash. I already tried at least 10 different profiles reading other issues with no success.

beaune33, 2022-09-14:
@brunohoff I would appreciate it very much if cloud cutter would help also for newer devices, as it works in your case. Unfortunately I dont see how I could help. Maybe https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/140 does, as it looks very similar.

brunohoff, 2022-09-14:
@beaune33, thanks, seems like a good idea on the surface. Do you know if there is a way to obtain these keys without opening the devices, soldering and reading the original firmware?

I mean, because this is an exploit that can run code on the device, maybe it`s possible to build an algorithm that obtains the keys on those different devices.

beaune33, 2022-09-14:
@brunohoff Unfortunately I have no idea how to obtain the keys, so I hope there is somebody else who could give us a clue. By the way, there is a new issue regarding SSL, which also seems to be the same problem: https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/141

kuba2k2, 2022-09-16:
The SSL keys are supposed to be generated & replaced by cloudcutter. There isn't a "list of keys" for a module/device - there's only a "profile" which is basically guiding Cloudcutter how to replace the keys.

SSL errors mean just that Cloudcutter failed to replace the keys - which is caused by a wrong profile being used.

beaune33, 2022-09-17:
@kuba2k2 Thanks for explaining what the SSL error means. In this case it means, that obviously there are two different versions of the LSC Moodlight in the market: the old one, we have a working profile for, and a new one which obviously is not supported now. So what is your proposal how to proceed? Is there anything I could try? Is it possible to manually modify the existing profile? What exactly needs to be discovered to create a new profile? Can this be done without opening the device? Thanks for your help!

kuba2k2, 2022-09-17:
To create a profile for a device we need a firmware dump. Currently there's no possibility to do it any other way. Additionally, if a device ever gets a software update from the vendor, it will very likely need a new profile.

Manually modifying profiles is not an option - there are two numbers (called offsets) that point to code functions in the firmware. Without a firmware dump, guessing these numbers is not possible at all.

closed 2022-11-30 by Cossid