tuya-cloudcutter/tuya-cloudcutter #126: WB2L - Arlec 12W 1050lm RGB Colour Mode Grid Connect Smart LED B22 Globe

brendantay, 2022-08-04:
This is a dump of an Arlec 12W 1050lm RGB Colour Mode Grid Connect Smart LED B22 Globe
WB2L chip.

https://www.bunnings.com.au/arlec-12w-1050lm-rgb-colour-mode-grid-connect-smart-led-b22-globe-3-pack_p0283331

This is a virgin bin dump:
[initial_dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9264043/initial_dump.zip)
 
This was a dump after connecting to the GRID App and controlling device etc.
[firmware.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9264046/firmware.zip)

Please let me know if anything else is required? :)

brendantay, 2022-08-04:
The above firmware was created using HID_download_py I have also created a dump using [https://github.com/khalednassar/bk7231tools](url)

The Dump (after connecting to Tuya Servers 
[initial_dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9264094/initial_dump.zip)

These are the dissected dump files [dissected_dump.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9264095/dissected_dump.zip)

After finding a reference to Haxomatic in another issue, I have run the dissected dump through it, Results below:

`[!] Loading and disassembling code - may take a moment
[+] Code loaded!
[!] Searching for post-vuln code patterns
[+] Found a post-vuln code pattern match!
[+] Matched instructions:

        0xb6784: adds r4, #0xfc
        0xb6786: ldr r3, [r4, #0x50]
        0xb6788: ldr r1, [pc, #0xb0]
        0xb678a: adds r0, r6, #0
[+] Identified lan object register as r4
[!] Searching for JSON object register
[+] Identified JSON object register as r7
[!] Searching for ty_cJSON_Parse function address
[+] ty_cJSON_Parse address: 0xc79b0
[!] Searching for mf_cmd_process gadget address
[+] mf_cmd_process gadget address (THUMB): 0xad515
[!] Searching for a mov r0, r7 intermediate gadget
[+] Found usable intermediate gadget at address 0xba65c:
        0xba65c: ldr r3, [r4]
        0xba65e: adds r0, r7, #0
        0xba660: blx r3
[+] Payload gadgets (THUMB): intermediate_gadget_addr=0xba65d mf_cmd_gadget_addr=0xad515`

bonuzzz, 2022-08-05:
@brendantay hi. what did you do with dump to prepare for hexomatic? thanks in advance

brendantay, 2022-08-05:
> @brendantay hi. what did you do with dump to prepare for hexomatic? thanks in advance

Simply dissected the dump using bk7231tools and ran hexomatic against the decrypted app code and it worked without issue. :)

bonuzzz, 2022-08-05:
ok. thanks. I missed dissect command in bk7231tools

ewand7, 2022-08-21:
I just cut the same model but with e27 using the other arlec profile. It seemed to work as the ap now begins with A- rather then grid- ,however i cannot seem to get openbeken to flash.

Cossid, 2022-10-15:
For reference, the Phopollo SBE2606 profile should work with this bulb.

closed 2022-11-30 by Cossid

Cossid, 2023-04-20:
@brendantay This device has an LED driver in it, do you know which LED driver it is? (usually one of: SM2135, BP5758D, BP1658CJ)