tuya-cloudcutter/tuya-cloudcutter #133: DPID Schema Extraction

narottamroyal, 2022-08-25:
Just wondering if anyone could explain how the schemas are stored in flash.

I found the SSID/password and some other JSON strings, and I can see that the data at `0x1EF000` changes after the device has connected. However, I can't quite figure out how the schemas are encoded... Are they stored as cJSON structs or something? How does the binary data correspond with the JSON files in the device profiles?

I've been finding this project super interesting, so any insight would be appreciated 🙂

kuba2k2, 2022-09-18:
The data stored in the upper flash blocks is encrypted using AES with a key derived from some constant strings AND an encrypted key stored in another block.

The encrypted part is a proprietary Tuya key-value storage format, that contains raw JSON data, including the UUID, AuthKey, PskKey as well as the firmware key and version numbers, and the data point descriptions.

So, having a dump should make it possible to not only find right code addresses, but also extract the device schema, without connecting to Tuya servers.

Note that the device schema is likely present after pairing the device first.

Cossid, 2022-12-01:
bk7231tools has been updated to include storage dumping.  If the device has a `user_param_key` array, pin mappings will be contained in that section (Most devices do, but not all.  It is possible [and likely] a device must be paired with Smart Life for that data to exist)

narottamroyal, 2022-12-02:
Thanks for the update! I just tested storage extraction on a dump I had lying around and it seemed to work well 🙂

It looks like most of the juicy details I was interested in can be found in [storage.py](https://github.com/tuya-cloudcutter/bk7231tools/blob/main/bk7231tools/analysis/storage.py), so I'll close this issue.

closed 2022-12-02 by narottamroyal