tuya-cloudcutter/tuya-cloudcutter #139: LSPA7 CB2S + BL0942

mario-almeida, 2022-09-12:
default without pairing
[firmware.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9550699/firmware.zip)

mario-almeida, 2022-09-12:
data points
```
{
  "result": [
    {
      "dpName": "Switch 1",
      "dpId": 1
    },
    {
      "dpName": "Relay Status",
      "dpId": 38
    },
    {
      "dpName": "Overcharge Switch",
      "dpId": 39
    },
    {
      "dpName": "Countdown 1",
      "dpId": 9
    },
    {
      "dpName": "Cycle Time",
      "dpId": 42
    },
    {
      "dpName": "Random Time",
      "dpId": 43
    },
    {
      "dpName": "Add Electricity",
      "dpId": 17
    },
    {
      "dpName": "Current",
      "dpId": 18
    },
    {
      "dpName": "Power",
      "dpId": 19
    },
    {
      "dpName": "Voltage",
      "dpId": 20
    },
    {
      "dpName": "Test Bit",
      "dpId": 21
    },
    {
      "dpName": "voltage coe",
      "dpId": 22
    },
    {
      "dpName": "electric coe",
      "dpId": 23
    },
    {
      "dpName": "power coe",
      "dpId": 24
    },
    {
      "dpName": "electricity coe",
      "dpId": 25
    },
    {
      "dpName": "Fault",
      "dpId": 26
    }
  ],
  "t": 1663008148282,
  "success": true,
  "status": "ok"
}
```

openshwprojects, 2022-09-12:
@mario-almeida hey, in general it's better to submit paired (but with dummy SSID) dumps

mario-almeida, 2022-09-13:
After pairing
[firmware_paired.zip](https://github.com/tuya-cloudcutter/tuya-cloudcutter/files/9556588/firmware_paired.zip)

closed 2022-11-30 by Cossid