tuya-cloudcutter/tuya-cloudcutter #140: allow software to cycle SSH Keys from other devices in the case that the device chosen doesn't work

ElGranPepo, 2022-09-13:
it looks like a lot of devices have different revisions and some products are repackaged by different manufacturers.
Asking for a feature that could search all keys that have been found for a particular module.

maybe an option to choose a module (wb2s, wb3s, or BK7231T or BK7231N instead of a product name

kuba2k2, 2022-09-16:
The SSL keys are supposed to be generated & replaced by cloudcutter. There isn't a "list of keys" for a module/device - there's only a "profile" which is basically guiding Cloudcutter how to replace the keys.

SSL errors mean just that Cloudcutter failed to replace the keys - which is caused by a wrong profile being used.

ElGranPepo, 2022-09-17:
> The SSL keys are supposed to be generated & replaced by cloudcutter. There isn't a "list of keys" for a module/device - there's only a "profile" which is basically guiding Cloudcutter how to replace [...]

can you give more clarity on this?  

if im running the script the first step i see is:

_Found access point name: "SmartLife-596B", trying to connect..
Device 'wlan0' successfully activated with '887248a6-365f-4278-9a7d-1ae6b657bf0e'.
Connected to access point.
Waiting 1 sec to allow device to set itself up..
Running initial exploit toolchain..
Exploit run, saved device config to!
output=/work/configured-devices/vLsK12qUrTwj.deviceconfig
Saved device config in /work/configured-devices/vLsK12qUrTwj.deviceconfig_

it says the exploit is run at this point correct?
Is there anything in the device config file that is created that can help us correlate the device to the correct profile at this point?

kuba2k2, 2022-09-17:
The created `.deviceconfig` file probably contains the generated SSL keys. Cloudcutter thinks the exploit succeeded, but in reality the device might have not saved the new keys at all - if a wrong profile is used.

The easiest way to verify this is to check the SSID. Cloudcutter prints the SSID it connects to. If it's always `SmartLife-596B` then the exploit didn't run. The SSID should begin with `A-` after a successful exploit run.

ElGranPepo, 2022-09-17:
> The created `.deviceconfig` file probably contains the generated SSL keys. Cloudcutter thinks the exploit succeeded, but in reality the device might have not saved the new keys at all - if a wrong [...]

is there any way to update the script to give an error if it doesnt see the correct SSID or notify in some way?  i think this is what is confusing most people. 

thank you for all your help!

kuba2k2, 2022-09-17:
There should be a way to do so. Someone who's experienced with cloudcutter's code could make these changes.

ElGranPepo, 2022-09-17:
i havent seen any updates for a while, not sure if the devs are still working on this project :(

ElGranPepo, 2022-09-18:
> 

the question is more of how many different devices there really are and firmware configs for these boards.  im not proposing to remove the manual choices, just add another option to cycle through them, this would help newbies and would provide a valid profile to use on multiple devices. Lets say i bought 4x 2 way switches or lightbulbs, i would run the script to go through all configs overnight to find a compatible config, then the next day do the other 3 devices with the config that works.  this would save me time and frustration while allowing me to be successful in moving my switches off the cloud.

kuba2k2, 2022-09-18:
That may work. But with the amount of profiles we have, compared to the amount of distinct devices that exist, chances of finding the right one are very low. As I said, there are millions of possible combinations of the offset numbers. 

If there is a matching profile already done, one can try profiles of similar devices. I.e. if you have an RGBCW bulb that's not yet supported, try profiles of other RGBCW bulbs.

Cossid, 2022-12-10:
Closing this as won't fix.  This isn't really feasible for a couple of reasons.  First there are now dozens of profiles.  Additionally, the SSL error is only known after attempting the exploit, and manual restarting of the device is needed each time.  There is no friendly way to automate this in such a way currently.

closed 2022-12-10 by Cossid