tuya-cloudcutter/tuya-cloudcutter #141: Detach fail after device connect to cloudcutter AP (SSL Error)

lukasita, 2022-09-13:
SmartLife device after connect to  cloudcutter AP but then SSL error appear and script stop. 

Any help what I can reconfigure

Device is 2 gang switch unknown vendor (from shoppe) .

 Building cloudcutter docker image..
Sending build context to Docker daemon  36.08MB
Step 1/10 : FROM python:3.9-slim-buster AS base
 ---> 457c189a27d2
Step 2/10 : RUN apt-get -qq update && apt-get install -qy --no-install-recommends git hostapd rfkill dnsmasq build-essential libssl-dev iproute2 mosquitto
 ---> Using cache
 ---> 4e21ff7889cb
Step 3/10 : FROM base AS python-deps
 ---> 4e21ff7889cb
Step 4/10 : RUN pip install --upgrade pipenv
 ---> Using cache
 ---> 22ec6ab958fc
Step 5/10 : COPY src/Pipfile /src/
 ---> Using cache
 ---> 1eca814c0a95
Step 6/10 : COPY src/Pipfile.lock /src/
 ---> Using cache
 ---> 80df66cc51ae
Step 7/10 : RUN cd /src && PIPENV_VENV_IN_PROJECT=1 pipenv install --deploy
 ---> Using cache
 ---> 5ce4de1e1b52
Step 8/10 : FROM python-deps AS cloudcutter
 ---> 5ce4de1e1b52
Step 9/10 : COPY src /src
 ---> Using cache
 ---> c4355395e8e1
Step 10/10 : WORKDIR /src
 ---> Using cache
 ---> 487855878a91
Successfully built 487855878a91
Successfully tagged cloudcutter:latest
Successfully built docker image
[?] Select the brand of your device: Tuya-Generic
   LEDVANCE
   Lightway
   Lonsonho
   LSC
   MartinJerry
   MINCO_HEAT
   Mlambert
   Mycket
   PHOPOLLO
   QIACHIP
   STITCH
   TreatLife
 > Tuya-Generic

[?] Select the article number of your device: WiFi-Switch-2-Gang
   E27-A60-RGBCWW-V1
   E27-RGBCW-Smart-Life-WB2L_M1
   LSPA9-Smart-Plug-BL0937
   WiFi-Curtain-Switch-WF-CS01
 > WiFi-Switch-2-Gang
   WiFi-Switch-3-Gang-CB3S
   WiFi-Switch-4-Gang

==> Toggle Tuya device's power off and on again 6 times, with ~1 sec pauses in between, to enable AP mode. Repeat if your device's SSID doesn't show up within ~30 seconds.
Wiping NetworkManager configs
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
Scanning for "A-" "Geeni-" "GRID-" "iHome-" "LDV SMART+-" "Merkury-" "Nexxt Home-" "SL-CreeLighting-" "SL-FLSNT-" "SmartLife-" "TreatLife-SL-" SSID...
[... 106 lines cut ...]
[W 220913 21:14:22 iostream:1406] SSL Error on 12 ('10.42.42.16', 53899): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Sep 13 21:14:23 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.16
Sep 13 21:14:23 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 220913 21:14:23 iostream:1406] SSL Error on 12 ('10.42.42.16', 53900): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Sep 13 21:14:24 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.16
Sep 13 21:14:24 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 220913 21:14:24 iostream:1406] SSL Error on 12 ('10.42.42.16', 53901): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Sep 13 21:14:26 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.16
Sep 13 21:14:26 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 220913 21:14:26 iostream:1406] SSL Error on 12 ('10.42.42.16', 53902): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Sep 13 21:14:27 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.16
Sep 13 21:14:27 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 220913 21:14:27 iostream:1406] SSL Error on 12 ('10.42.42.16', 53903): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Sep 13 21:14:28 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.16
Sep 13 21:14:28 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 220913 21:14:28 iostream:1406] SSL Error on 12 ('10.42.42.16', 53904): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)
Sep 13 21:14:29 dnsmasq[15]: query[A] h3.iot-dns.com from 10.42.42.16
Sep 13 21:14:29 dnsmasq[15]: config h3.iot-dns.com is 10.42.42.1
Using PSK v1 - Received PSK ID version 02
[W 220913 21:14:29 iostream:1406] SSL Error on 12 ('10.42.42.16', 53905): [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1129)

lukasita, 2022-09-13:
Additionally I found that this error message originate from src/cloudcutter/crypto/pskcontext.py

kuba2k2, 2022-09-16:
The SSL keys are supposed to be generated & replaced by cloudcutter. There isn't a "list of keys" for a module/device - there's only a "profile" which is basically guiding Cloudcutter how to replace the keys.

SSL errors mean just that Cloudcutter failed to replace the keys - which is caused by a wrong profile being used.

Cossid, 2022-10-15:
The profile you used did not work for this device, as your AP prefix did not change to start wtih A-
What device is this?  What Tuya module is inside?  What Tuya software version does it have?

Cossid, 2022-12-10:
Closing as stale.  Please provide additional information if you re-open this issue.

closed 2022-12-10 by Cossid